arduino/arduino-cli #2602: feat: reference sketch.yaml platforms for "upload", "board list", etc.

egnor, 2024-05-13:
### Describe the request

The [sketch project file `sketch.yaml`](https://arduino.github.io/arduino-cli/0.29/sketch-project-file/) is wonderful for making more self contained projects that don't include the dreaded "okay now here follow these 20 steps to Set Up Your Machine, by the way they're out of date" disease. (Version managers like `asdf` or `mise` are [another part of the solution](https://github.com/egnor/asdf-arduino-cli).)

BUT it only really has much effect for `arduino-cli compile`. When for example doing `arduino-cli upload`, if the core in question hasn't been installed with `arduino-cli core install`, the upload will often fail. Similarly, `arduino-cli board list` won't detect boards unless the core has been installed.

The `upload` command does parse `sketch.yaml` to find attachment information, so, there's some precedent for looking for one...

### Describe the current behavior

A core that is referenced in `sketch.yaml` but not globally installed on the system doesn't inform `arduino-cli upload` or `arduino-cli board list` commands.

### Arduino CLI version

arduino-cli  Version: 0.35.3 Commit: 95cfd654 Date: 2024-02-19T13:24:24Z

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

cmaglie, 2024-10-22:
@egnor Do you have an example that shows the problem? the `upload` command has a flag to specify the profile to use:

```
  -m, --profile string                Sketch profile to use
```

> Similarly, `arduino-cli board list` won't detect boards unless the core has been installed.

Uhm, that's a bit convoluted use case, let's suppose that we already have a `-m profile` flag in `board list` command: what should be the expected result, for example, if we use a profile that uses the `arduino:avr` platform? `board list -m profile_avr` should detect only the boards listed in the AVR platform and display the others as unknown?

egnor, 2024-10-22:
For `upload`, I think this actually got fixed at some point? I've been using the flag for a while now, and hadn't realized the bug hadn't been tagged. Maybe just as a part of some refactor?

Let's close this bug; the `board list` case is kind of its own complicated situation with other bugs talking about it.

closed 2024-10-22 by egnor