arduino/arduino-cli #2607: arduino-cli compile gives "Error during build: Invalid FQBN: not an FQBN: esp32:esp32:esp32c6:DebugLevel=verbose:CDCOnBoot=cdc:JTAGAdapter=builtin" error

Sfinx, 2024-05-16:
### Describe the problem

Error is:

```
arduino-cli compile -v -b esp32:esp32:esp32c6:DebugLevel=verbose:CDCOnBoot=cdc:JTAGAdapter=builtin sketch_hello
Error during build: Invalid FQBN: not an FQBN: esp32:esp32:esp32c6:DebugLevel=verbose:CDCOnBoot=cdc:JTAGAdapter=builtin
```

Details:

```
arduino-cli  Version: 0.35.3 Commit: 95cfd654 Date: 2024-02-19T13:24:24Z

arduino-cli board details --fqbn esp32:esp32:esp32c6
Board name:            ESP32C6 Dev Module
FQBN:                  esp32:esp32:esp32c6
Board version:         3.0.0-rc3

Package name:          esp32
Package maintainer:    Espressif Systems
Package URL:           https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json
Package website:       https://github.com/espressif/arduino-esp32
Package online help:   http://esp32.com

Platform name:         esp32
Platform category:     ESP32
Platform architecture: esp32
Platform URL:          https://github.com/espressif/arduino-esp32/releases/download/3.0.0-rc3/esp32-3.0.0-rc3.zip
Platform file name:    esp32-3.0.0-rc3.zip
Platform size (bytes): 23906520
Platform checksum:     SHA-256:87ff8e3499c8c2112c3d6faf0017998e64135c0f3553088d76be6f8b6a5af1ef

Required tool: arduino:dfu-util                                  0.11.0-arduino5
Required tool: esp32:esp-rv32                                    2302
Required tool: esp32:esp32-arduino-libs                          idf-release_v5.1-442a798083
Required tool: esp32:esptool_py                                  4.6
Required tool: esp32:mklittlefs                                  3.0.0-gnu12-dc7f933
Required tool: esp32:mkspiffs                                    0.2.3
Required tool: esp32:openocd-esp32                               v0.12.0-esp32-20230921
Required tool: esp32:riscv32-esp-elf-gdb                         12.1_20231023
Required tool: esp32:xtensa-esp-elf-gdb                          12.1_20231023
Required tool: esp32:xtensa-esp32-elf-gcc                        esp-12.2.0_20230208
Required tool: esp32:xtensa-esp32s2-elf-gcc                      esp-12.2.0_20230208
Required tool: esp32:xtensa-esp32s3-elf-gcc                      esp-12.2.0_20230208

Option:        Upload Speed                                                                  UploadSpeed
               921600                                            ✔                           UploadSpeed=921600
               115200                                                                        UploadSpeed=115200
               230400                                                                        UploadSpeed=230400
               460800                                                                        UploadSpeed=460800
Option:        USB CDC On Boot                                                               CDCOnBoot
               Disabled                                          ✔                           CDCOnBoot=default
               Enabled                                                                       CDCOnBoot=cdc
Option:        CPU Frequency                                                                 CPUFreq
               160MHz (WiFi)                                     ✔                           CPUFreq=160
               120MHz (WiFi)                                                                 CPUFreq=120
               80MHz (WiFi)                                                                  CPUFreq=80
               40MHz                                                                         CPUFreq=40
               20MHz                                                                         CPUFreq=20
               10MHz                                                                         CPUFreq=10
Option:        Flash Frequency                                                               FlashFreq
               80MHz                                             ✔                           FlashFreq=80
               40MHz                                                                         FlashFreq=40
Option:        Flash Mode                                                                    FlashMode
               QIO                                               ✔                           FlashMode=qio
               DIO                                                                           FlashMode=dio
Option:        Flash Size                                                                    FlashSize
               4MB (32Mb)                                        ✔                           FlashSize=4M
               8MB (64Mb)                                                                    FlashSize=8M
[... 51 lines cut ...]
### Expected behavior

compilation should succeed

### Arduino CLI version

arduino-cli  Version: nightly-20240516 Commit: 8b78c05 Date: 2024-05-16T01:30:15Z

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Additional context

Single option works, i.e. "-b esp32:esp32:esp32c6:DebugLevel=verbose". Seems like cli fails parsing options > 1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

Sfinx, 2024-05-16:
it is appeared that comma must be used as option separator - my fault

per1234, 2024-05-16:
Hi @Sfinx. The format of the FQBN is like this:

```text
<vendor ID>:<architecture>:<board ID>[:<menu ID>=<option ID>[,<menu ID>=<option ID>]...]
```

So the correct FQBN would be:

```text
esp32:esp32:esp32c6:DebugLevel=verbose,CDCOnBoot=cdc,JTAGAdapter=builtin
```

(note that `,` is used as the separator of multiple custom board options, NOT `:`)

---

If you find it inconvenient to set custom board options via the FQBN, you might prefer using `--board-options` flags instead:

https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_compile/#options

> ```text
> --board-options strings                 List of board options separated by commas. Or can be used multiple times for multiple options.
> ```

So you can use this command:

```text
arduino-cli compile -b esp32:esp32:esp32c6:DebugLevel=verbose,CDCOnBoot=cdc,JTAGAdapter=builtin
```

or this:

```text
arduino-cli compile -b esp32:esp32:esp32c6 --board-options DebugLevel=verbose --board-options CDCOnBoot=cdc --board-options JTAGAdapter=builtin
```

closed 2024-05-16 by per1234