arduino/arduino-cli #2613: changing the name of src directory, don't change compilator cache

FroggyCorp, 2024-05-26:
### Describe the problem

Hello everyone,

I have files like :

`myfile.ino`
`src/mylib.cpp`
`src/mylib.h`

I get an error at compile time on `mylib.h`:

```text
[...]\myfile\src\mylib.h:28:55: error: 'TOTO' was not declared in this scope
```

In fact, `mylib.cpp` is not include in `myfile`, because i don't find anyway to avoid compile of `mylib.cpp`, I just changed the name of directory from `src` to `src2`, so my files are now:

`myfile.ino`
`src2/mylib.cpp`
`src2/mylib.h`

But the error is still the same with a wrong directory which doesn't exist anymore (`\myfile\src\mylib.h`):

```text
[...]\myfile\src\mylib.h:28:55: error: 'TOTO' was not declared in this scope
```

The fault occurs when the files are under a subfolder that contains any name starting with `s`.

The fault does not occur after I change the folder name to something that does not start with `s`.

### To reproduce

1. Create a sketch.
1. Create `src` directory inside.
1. Create some library inside `src/` which produces a compile error.
1. Change name of `src/` to `s/`.
1. Compile the sketch again.

### Expected behavior

Checking that files still exists in path

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-05-26:
Hi @FroggyCorp. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/1240.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

---

You can follow the instructions at https://github.com/arduino/arduino-cli/issues/1240#user-content-workaround-ide to workaround the bug.

> The fault does not occur after I change the folder name to something that does not start with s.

I think the fact the fault no longer occurred after you did that was purely coincidental. I tested it myself and the fault still occurs for me even after renaming the folder to something that does not start with `s`. My guess is that you saved the sketch to a different name at the same time. Doing that causes a new temporary build folder to be used.

closed 2024-05-26 by per1234

FroggyCorp, 2024-05-27:
Hey,

No problem :)