arduino/arduino-cli #2616: Use of {runtime.platform.path} makes referencing other cores often impossible

egnor, 2024-05-28:
### Describe the problem

(I found this issue discussed various places, but no specific open bug for the problem?)

The ability to [reference other cores from a platform](https://arduino.github.io/arduino-cli/dev/platform-specification/#referencing-another-core-variant-or-tool) seems to be an important design feature, and it is [the recommended way to create custom `boards.txt` entries](https://github.com/arduino/arduino-cli/issues/1684) without having to stuff every board ever made into the platform vendor's `boards.txt` file.

HOWEVER, in practice, for many platforms (including very popular ones like rp2040 and esp32), this referencing doesn't work. The reason is that those platform definitions reference `{runtime.platform.path}` heavily. For example, the [arduino-esp32 `platform.txt` file](https://github.com/espressif/arduino-esp32/blob/master/platform.txt) does this sort of thing

```
...
tools.esp32-arduino-libs.path={runtime.platform.path}/tools/esp32-arduino-libs
...
compiler.sdk.path={tools.esp32-arduino-libs.path}/{build.mcu}
...
compiler.cpreprocessor.flags="@{compiler.sdk.path}/flags/defines" "-I{build.source.path}" -iprefix "{compiler.sdk.path}/include/" "@{compiler.sdk.path}/flags/includes" "-I{compiler.sdk.path}/{build.memory_type}/include"
...
```

Many of the uses are deep in long, complex operations one wouldn't want to cut-and-paste:

```
recipe.hooks.prebuild.4.pattern=bash -c "[ -f "{build.source.path}"/bootloader.bin ] && cp -f "{build.source.path}"/bootloader.bin "{build.path}"/{build.project_name}.bootloader.bin || ( [ -f "{build.variant.path}"/{build.custom_bootloader}.bin ] && cp "{build.variant.path}"/{build.custom_bootloader}.bin "{build.path}"/{build.project_name}.bootloader.bin || cp -f "{runtime.platform.path}"/tools/sdk/{build.mcu}/bin/bootloader_{build.boot}_{build.boot_freq}.bin "{build.path}"/{build.project_name}.bootloader.bin )"
```

One might say this is a bug in arduino-esp32 (and arduino-pico, and...), and they should use `{build.core.path}` instead. And maybe so-- but if they do, their platform.txt becomes incompatible with older versions of Arduino, e.g. before https://github.com/arduino/arduino-cli/issues/1012 was fixed.

Speaking of https://github.com/arduino/arduino-cli/issues/1012, the discussion there points to use of symlinks as a workaround, but that's clearly messy, difficult, fragile, and hard to automate. There's also discussion of this issue in https://github.com/esp8266/Arduino/issues/5769.

Until this is solved, external core reference and all the reuse it was supposed to enable is basically broken.

### To reproduce

Attempt to reference another core from a simple boards-only platform definition, as recommended in various places

### Expected behavior

Core reference works, code builds and uploads

### Arduino CLI version

arduino-cli  Version: 0.35.3 Commit: 95cfd654 Date: 2024-02-19T13:24:24Z

### Operating system

N/A

### Operating system version

All

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

egnor, 2024-05-28:
Proposal:
- if `[board].use_core_platform=true`, set `{runtime.platform.path}` to the CORE platform path (otherwise set it to the BOARD platform path as current)
- ALWAYS set `{runtime.board.platform.path}` to the BOARD platform path
- ALWAYS set `{runtime.core.platform.path}` to the CORE platform path

Also best fix [this](https://arduino.github.io/arduino-cli/dev/platform-specification/#burn-bootloader):
> When using the Arduino IDE, if the board uses a [core reference](https://arduino.github.io/arduino-cli/dev/platform-specification/#core-reference), the platform.txt of the [core platform](https://arduino.github.io/arduino-cli/dev/platform-specification/#platform-terminology) is not used at all in defining the recipes for erase and bootloader actions. When using Arduino development software other than the Arduino IDE, the handling of properties from the core platform's platform.txt is done as usual.

This means
- no changes to default behavior
- no changes needed to existing platforms
- doesn't break existing external-core-using boards if they work
- enables new external-core-using boards to work with all platforms (I think)

cmaglie, 2024-06-17:
@egnor thank you very much for the valuable bug report.

Can you provide a POC core that references an esp32/pico core (to demonstrate the issue)?
I want to work on a fix, and your proposal looks like a good solution, I need something to test on and I can make up a test, but if you already have a good example it would be much better.

egnor, 2024-06-17:
(will do, stay tuned!)

egnor, 2024-06-18:
OK here's a minimal example: https://github.com/egnor/repro_arduino_cli_2616

1. Install `arduino-cli` v1.0.0
2. `git clone https://github.com/egnor/repro_arduino_cli_2616`
3. `cd repro_arduino_cli_2616/test_sketch`
4. `arduino-cli compile`

There's a `sketch.yaml` file there that defines the platform URLs, including the base Espressif platform and also that repo itself (on github). When I try, I get:

```
Downloading platform esp32:esp32@3.0.1 (https://espressif.github.io/arduino-esp32/package_esp32_index.json)...
Downloading index: package_index.tar.bz2 downloaded                             
Downloading index: package_esp32_index.json downloaded                          
esp32:esp32@3.0.1 downloaded                                                    
Installing platform esp32:esp32@3.0.1...
Downloading tool esp32:esp32-arduino-libs@idf-release_v5.1-442a798083...
esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 23.59 MiB / 356.11 MiB    6esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 51.31 MiB / 356.11 MiB   14esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 79.44 MiB / 356.11 MiB   22esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 107.28 MiB / 356.11 MiB   3esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 135.06 MiB / 356.11 MiB   3esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 135.06 MiB / 356.11 MiB   3esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 163.00 MiB / 356.11 MiB   4esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 190.76 MiB / 356.11 MiB   5esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 218.73 MiB / 356.11 MiB   6esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 246.37 MiB / 356.11 MiB   6esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 246.37 MiB / 356.11 MiB   6esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 274.28 MiB / 356.11 MiB   7esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 302.04 MiB / 356.11 MiB   8esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 329.86 MiB / 356.11 MiB   9esp32:esp32-arduino-libs@idf-release_v5.1-442a798083 downloaded                 
Installing tool esp32:esp32-arduino-libs@idf-release_v5.1-442a798083...
Downloading tool esp32:esp-x32@2302...
esp32:esp-x32@2302 downloaded                                                   
Installing tool esp32:esp-x32@2302...
Downloading tool esp32:esp-xs2@2302...
esp32:esp-xs2@2302 downloaded                                                   
Installing tool esp32:esp-xs2@2302...
Downloading tool esp32:esp-xs3@2302...
esp32:esp-xs3@2302 downloaded                                                   
Installing tool esp32:esp-xs3@2302...
Downloading tool esp32:xtensa-esp-elf-gdb@12.1_20231023...
esp32:xtensa-esp-elf-gdb@12.1_20231023 downloaded                               
Installing tool esp32:xtensa-esp-elf-gdb@12.1_20231023...
Downloading tool esp32:esp-rv32@2302...
esp32:esp-rv32@2302 downloaded                                                  
Installing tool esp32:esp-rv32@2302...
Downloading tool esp32:riscv32-esp-elf-gdb@12.1_20231023...
esp32:riscv32-esp-elf-gdb@12.1_20231023 downloaded                              
Installing tool esp32:riscv32-esp-elf-gdb@12.1_20231023...
Downloading tool esp32:openocd-esp32@v0.12.0-esp32-20230921...
esp32:openocd-esp32@v0.12.0-esp32-20230921 downloaded                           
Installing tool esp32:openocd-esp32@v0.12.0-esp32-20230921...
Downloading tool esp32:esptool_py@4.6...
esp32:esptool_py@4.6 downloaded                                                 
Installing tool esp32:esptool_py@4.6...
Downloading tool esp32:mkspiffs@0.2.3...
esp32:mkspiffs@0.2.3 downloaded                                                 
Installing tool esp32:mkspiffs@0.2.3...
Downloading tool esp32:mklittlefs@3.0.0-gnu12-dc7f933...
esp32:mklittlefs@3.0.0-gnu12-dc7f933 downloaded                                 
Installing tool esp32:mklittlefs@3.0.0-gnu12-dc7f933...
Downloading tool arduino:dfu-util@0.11.0-arduino5...
arduino:dfu-util@0.11.0-arduino5 downloaded                                     
Installing tool arduino:dfu-util@0.11.0-arduino5...
Warning: Board egnor:esp32:derived_esp32 doesn't define a 'build.board' preference. Auto-set to: ESP32_DERIVED_ESP32
cp: cannot stat '/home/egnor/source/repro_arduino_cli_2616/arduino_data/internal/egnor_esp32_1.0.1_37f3eb13a64228ba/tools/partitions/{build.partitions}.csv': No such file or directory
```

(I'm ignoring the warning, probably I should have defined `build.board` also, but the auto-setting is fine.)

The `cp` doesn't work because of this line (and ones around it) in the base esp32 `platform.txt`, which uses `{runtime.platform.path}` as described in the initial bug report:
https://github.com/espressif/arduino-esp32/blob/99750cd37e4bef0875d36cff7e319f327d8532a2/platform.txt#L113

(I think there might also be an issue with `{build.partitions}`, which [should be inherited from the base esp32 `boards.txt`](https://github.com/espressif/arduino-esp32/blob/99750cd37e4bef0875d36cff7e319f327d8532a2/boards.txt#L1189), but maybe that's just an output thing. In any case the directory is wrong.)

I have found by experience if I fudge my way through this particular issue, there are many others, as `{runtime.platform.path}` is used extensively in the base `platform.txt` definitions.

cmaglie, 2024-06-25:
Thanks @egnor
I've made up a first try in #2652, but the compile still fails. Probably other variables are missing.
I've slightly changed your boards.txt by adding the following lines:

```
derived_esp32.build.partitions=default_8MB
derived_esp32.build.mcu=esp32s3
derived_esp32.build.use_core_platform_for_runtime_platform_path=true
```

I'm sure that many other definitions are missing. I'm also a bit concerned this system could be a bit fragile since a change in the esp32 platform.txt (variable names, or how they are used...) could lead to a breaking change in the platforms that use esp32 as a reference.

egnor, 2024-06-26:
This is exciting @cmaglie !!

You're right, while the _platform_ is referenced, nothing about the _board_ definition is. I have updated my repo (https://github.com/egnor/repro_arduino_cli_2616) accordingly:
- made `derived_esp32` a full board definition, taking all the default options from the `esp32` board
- added an `update_package.py` that bumps the version, rebuilds the zipfile, edits the index file, and edits the sketch project file to all be consistent, for quicker iteration

You can see if that works? I'm going to try to pull down your PR branch and see if it's easy for me to build.

As far as fragility based on the platform definition -- a certain amount of that is inevitable with configuration extension like this BUT I agree that it feels like a lot of duplication. (Less, at least, than forking the entire platform!) I would propose some mechanism for "inheriting" one board definition from another, which I think is outside the scope of this particular piece of work. (It should be simpler than core reuse, since boards don't have associated assets, it's just a matter of copying all the key values and then overriding.)

egnor, 2024-06-26:
I also had to make sure to reference the variant externally:
```
derived_esp32.build.variant=esp32:esp32
```
having done so :tada: it works :tada: using your PR branch!
```
% ../../arduino-cli/arduino-cli compile    
Downloading platform egnor:esp32@1.0.4 (https://github.com/egnor/repro_arduino_cli_2616/raw/main/package_egnor_repro2616_index.json)...
Downloading index: package_index.tar.bz2 downloaded                             
Downloading index: package_egnor_repro2616_index.json downloaded                
egnor:esp32@1.0.4 downloaded                                                    
Installing platform egnor:esp32@1.0.4...
Sketch uses 275041 bytes (20%) of program storage space. Maximum is 1310720 bytes.
Global variables use 17008 bytes (5%) of dynamic memory, leaving 310672 bytes for local variables. Maximum is 327680 bytes.

Used platform Version Path
egnor:esp32   1.0.4   /home/egnor/source/repro_arduino_cli_2616/arduino_data/internal/egnor_esp32_1.0.4_708e4ca792f7a01c
esp32:esp32   3.0.1   /home/egnor/source/repro_arduino_cli_2616/arduino_data/internal/esp32_esp32_3.0.1_feaea63fd96e67bc
```

or at least the compile succeeded, I haven't tried running it on a board, but I think this is most of the way for sure!

I am excited to see this PR land, it will be a great step forward. (And then we can get board-to-board inheritance... and maybe some way to reference local index and/or package files without hitting a public URL... and standardize on some way to add compiler defines... then we're golden!!)

closed 2024-07-26 by cmaglie