arduino/arduino-cli #2633: Unable to upload when two identical Arudino Nano ESP32 devices attached

TallestHead, 2024-06-12:
### Describe the problem

When I have two Arduino Nano ESP32's attached to my Mac, I am unable to upload via `arduino-cli upload`. The `dfu-util` cannot figure out which device is which without a `--serial` argument and I cannot seem to supply one.

### To reproduce

I have two Nano ESP32s attached to my mac. Attempting to upload to one via the `-p` parameter leaves `dfu-util` confused.

```
reiman@notch USB-mapper % arduino-cli upload -p /dev/cu.usbmodemECDA3B6 --fqbn arduino:esp32:nano_nora  HostSide
dfu-util 0.11-arduino4

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device

Failed uploading: uploading error: exit status 74
```

### Expected behavior

Ideally, `upload` could handle this or `dfu-util` could. If handling this magically is too difficult, allow users to pass arbitrary arguments to `dfu-util` or at least support a `--seralno` argument and pass that down as `--serial`. 

### Arduino CLI version

arduino-cli  Version: nightly-20240612 Commit: 5c5c176 Date: 2024-06-12T01:32:30Z

### Operating system

macOS

### Operating system version

14.5

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

TallestHead, 2024-06-12:
I tested both nightly and release with the same result.

ednieuw, 2025-05-22:
Still an issue

per1234, 2025-05-22:
Arduino CLI already makes the serial number of the selected port available for use by any Arduino boards platform:

https://arduino.github.io/arduino-cli/dev/platform-specification/#properties-from-pluggable-discovery

This makes it possible to handle this magically, which is superior to the alternative proposal for putting the burden of providing the serial number on the user.

So I will close this as resolved.

As for resolving the problem for the Nano ESP32 board, it only requires that the "Arduino ESP32 Boards" platform uses the serial number information provided by Arduino CLI in the upload command pattern that is defined by the `platform.txt` configuration file. That has been done:

https://github.com/arduino/arduino-esp32/commit/2d3ceda2d57e0f95598fc7ae6e0bcaab67da0c61

However, that work was done after the time of the last release of the platform (2.0.18-arduino.5). So the problem will still occur when using platform installed via `arduino-cli core install` or Arduino IDE's Boards Manager. For now, you can manually make the change from the commit I linked above your local installation of the platform, or you can wait for the next release of the platform to come out.

I provided detailed instructions for manually making the necessary change over on Arduino Forum:

https://forum.arduino.cc/t/cant-upload-to-more-than-1-nano-esp32-dfu-util-more-than-one-dfu-capable-usb-device-found/1162959/9

closed 2025-05-22 by per1234