arduino/arduino-cli #2638: Schema not defined for key 'sketchbook_path'

brightproject, 2024-06-14:
### Describe the problem

I used version `0.35.3`
Updated to **1.0.0**...
Now the utility’s performance has become poor during compilation, as it actually writes with any command
`schema not defined for key 'sketchbook_path'`
Compilation seems to work, but it takes much longer than before - they work in Windows.
Port monitor command

> PS C:\Windows\system32> arduino-cli monitor -p COM12 -c baudrate=230400

displays this error
`schema not defined for key 'sketchbook_path'`
Please specify an FQBN. The board on port COM12 with protocol serial can't be identified
The download also seems to work, but I don't understand what has changed and how to fix it.
When updating software there is always a chance to break something, and you guys managed to do it.
Reverted to stable version `0.35.3`

### To reproduce

Command:

> arduino-cli compile -b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE C:\Users\Documents\Arduino\libraries\Test\Test.ino

```
schema not defined for key 'sketchbook_path'
```

> arduino-cli version

```
schema not defined for key 'sketchbook_path'
arduino-cli  Version: 1.0.0 Commit: 05c9852a Date: 2024-06-12T14:13:22Z
```

### Expected behavior

Without comments

### Arduino CLI version

1.0.0

### Operating system

Windows

### Operating system version

10 Pro

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2024-06-14:
You probably have the key `sketchbook_path` defined in your `arduino-cli.yml` configuration file.
That key is incorrect, the correct key is `directories.user`, for example in my `arduino-cli.yaml`:
```
directories:
  user: /home/cmaglie/Arduino
```

The directory set in `sketchbook_path` is ignored by the CLI, the message is just a warning.

cmaglie, 2024-06-14:
> Port monitor command
> 
> > PS C:\Windows\system32> arduino-cli monitor -p COM12 -c baudrate=230400
> 
> displays this error
> 
> ```
> schema not defined for key 'sketchbook_path'
> Please specify an FQBN. The board on port COM12 with protocol serial can't be identified
> ```

In this case the error is telling that you must specify the board with `-b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE`, because it can not be automatically detected via USB vid/pid.

The board is required to determine the initial settings for RTS/DTR.

cmaglie, 2024-06-14:
> When updating software there is always a chance to break something, and you guys managed to do it.

Breaking changes are to be expected until the CLI was in alpha/beta state. The version number `0.x.x` is for that starting period where the API is not well defined and it's subject to breaking changes as we refine the user interface.

To reduce the impact on the users we decided to put all the breaking changes on the 1.0.0 release now. After this release, it can be considered stable and other breaking changes should not be expected anymore until the 2.0.0. Of course, there may be bugs, or unexpected changes from users coming from 0.x.x, but hopefully this should but the last time we involve our users into unexpected breaking...

BTW, the error message could be surely improved, so thanks for your bug report and your patience :-)

brightproject, 2024-06-14:
> BTW, the error message could be surely improved, so thanks for your bug report and your patience :

I expected to use the utility as before after the update, i.e. in the usual way.
But as a result, do I need to configure something, learn new commands and flags?
I'm not outraged, I'd just like to understand how the update was carried out...
It is surprising that earlier versions were minor and indeed no problems were visible, although the operation of the utility was improved from version to version.
Now there is a global change and immediately to `1.0.0` and it turns out that the paradigm of using the utility has changed?

cmaglie, 2024-06-14:
> This issue tracker is only to be used to report bugs or feature requests specific to the project. Support requests must be made on Arduino Forum. I'm sure we will be able to help you out over there:

Yes, thanks @per1234.

BTW, besides the other requests for help, I confirm that the `monitor` command has indeed changed behavior, the `-b` wasn't required in 0.35.3. I'll check this in the coming days.

brightproject, 2024-06-17:
Hello @cmaglie 
To eliminate the appearance of the inscription during compile code

> schema not defined for key 'sketchbook_path'

I just commented out the line

> sketchbook_path: /home/User/Arduino

in the `arduino-cli.yaml file`

cmaglie, 2024-06-17:
Reopen because not fixed completely yet.

closed 2024-06-24 by cmaglie