arduino/arduino-cli #2646: Space-separated $ARDUINO_BOARD_MANAGER_ADDITIONAL_URLS broken in arduino-cli v1.0.0

egnor, 2024-06-18:
### Describe the problem

The environment variable `$ARDUINO_BOARD_MANAGER_ADDITIONAL_URLS` is [documented to allow a space-separated list of URLs](https://arduino.github.io/arduino-cli/0.36/configuration/#example_1), but as of v1.0.0 it does not.

### To reproduce

1. Install arduino-cli v1.0.0.
2. Use the example: `export ARDUINO_BOARD_MANAGER_ADDITIONAL_URLS="https://downloads.arduino.cc/packages/package_staging_index.json https://downloads.arduino.cc/packages/package_mbed_index.json"`
3. Run: `arduino-cli core update-index`

### Expected behavior

I expect it to load both package indexes. Instead, I get:
```
Downloading index: package_index.tar.bz2 downloaded                             
Downloading index: package_mbed_index.json Server responded with: 404 Not Found 
Some indexes could not be updated.
```

### Arduino CLI version

1.0.0

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Additional context

Running with `-v`, it looks like it's overescaping the value:
```
INFO[0000] arduino-cli version 1.0.0                    
INFO[0000] Using config file: /home/egnor/source/repro_arduino_cli_2616/arduino_data/arduino-cli.yaml 
INFO[0000] Executing `arduino-cli core update-index`    
INFO[0000] Checking signature                            index=/home/egnor/source/repro_arduino_cli_2616/arduino_data/package_index.json signatureFile=/home/egnor/source/repro_arduino_cli_2616/arduino_data/package_index.json.sig trusted=true
INFO[0000] Checking signature                            index=/home/egnor/source/repro_arduino_cli_2616/arduino_data/package_mbed_index.json signatureFile=/home/egnor/source/repro_arduino_cli_2616/arduino_data/package_mbed_index.json.sig trusted=true
INFO[0000] Loading hardware from: /home/egnor/source/repro_arduino_cli_2616/arduino_data/packages 
INFO[0000] Loading package builtin from: /home/egnor/source/repro_arduino_cli_2616/arduino_data/packages/builtin 
INFO[0000] Checking existence of 'tools' path: /home/egnor/source/repro_arduino_cli_2616/arduino_data/packages/builtin/tools 
INFO[0000] Loading tools from dir: /home/egnor/source/repro_arduino_cli_2616/arduino_data/packages/builtin/tools 
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loaded tool                                   tool="builtin:dfu-discovery@0.1.2"
INFO[0000] Loaded tool                                   tool="builtin:mdns-discovery@1.0.9"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.4.1"
INFO[0000] Loaded tool                                   tool="builtin:serial-monitor@0.14.1"
INFO[0000] Loading libraries index file                  index=/home/egnor/source/repro_arduino_cli_2616/arduino_data/library_index.json
INFO[0000] Adding libraries dir                          dir=/home/egnor/Arduino/libraries isSingleLibrary=false location=user
INFO[0000] Updating index                                url="https://downloads.arduino.cc/packages/package_index.tar.bz2"
INFO[0000] Starting download                             url="https://downloads.arduino.cc/packages/package_index.tar.bz2"
Downloading index: package_index.tar.bz2 downloaded                             
INFO[0000] Updating index                                url="https://downloads.arduino.cc/packages/package_staging_index.json%20https://downloads.arduino.cc/packages/package_mbed_index.json"
INFO[0000] Starting download                             url="https://downloads.arduino.cc/packages/package_staging_index.json%20https://downloads.arduino.cc/packages/package_mbed_index.json"
Downloading index: package_mbed_index.json Server responded with: 404 Not Found 
Some indexes could not be updated.
```

Note the URL `https://downloads.arduino.cc/packages/package_staging_index.json%20https://downloads.arduino.cc/packages/package_mbed_index.json` which is actually both URLs combined with a `%20` (URL-escaped space).

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-06-19:
Closing as duplicate of https://github.com/arduino/arduino-cli/issues/2643.

@egnor this bug has already been fixed: https://github.com/arduino/arduino-cli/pull/2645. The fix is available when using the latest nightly build of Arduino CLI.

You can download the nightly build from the links listed here:

https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds

egnor, 2024-06-19:
Oops shoulda paid more attention to the "tested on nightly" checkbox. My bad sorry!