arduino/arduino-cli #2658: arduino-cli v1.0.x breaks arduino/compile-sketches action for Teensy boards

tfry-git, 2024-07-07:
### Describe the problem

Around early June, the arduino/compile-sketches-action (https://github.com/arduino/compile-sketches/) fails for the Teensy core (https://www.pjrc.com/teensy/package_teensy_index.json). The error message is:

The compilation itself seems to work, as before, but always ends with:
> Opening Teensy Loader...
> Unable find Teensy Loader.  (p)  Is the Teensy Loader application running?
> quit

Both the above message, and the unsuccessful exit are new.

Testing the various components involved, this is neither due to updates in compile-sketches, nor it the core. However, downgrading arduino-cli to v0.35.3 fixes the issue.

It should be noted that an upload is absolutely not required at this point. We're essentially just running ```arduino-cli compile```, here.

Further resources:
- Example of a successful run in the Mozzi library (June 9th): https://github.com/sensorium/Mozzi/actions/runs/9438094175/job/259947127
- Example of a failed run in the Mozzi library (June 12th): https://github.com/sensorium/Mozzi/actions/runs/9488741406/job/26270443137
- Discussion on the Teensy forum: https://forum.pjrc.com/index.php?threads/teensy-boards-starting-to-fail-in-github-actions-arduino-compile-sketches.75247/

### To reproduce

This probably needs to be run inside a github-action, where the Teensy loader is not available.

An example workflow shall contain:
```
steps:
      - name: Checkout repository
        uses: actions/checkout@v4

      - name: Compile examples
        uses: arduino/compile-sketches@v1
        with:
          cli-version: 0.35.3  # may or may not be given, to toggle triggering the bug
          github-token: ${{ secrets.GITHUB_TOKEN }}
          fqbn: teensy:avr:teensy36
          platforms: |
              - name: teensy:avr
                source-url: https://www.pjrc.com/teensy/package_teensy_index.json
          sketch-paths: |
            - examples
```

### Expected behavior

```arduino-cli compile``` should compile, only, not attempt to contact the upload daemon.

### Arduino CLI version

v1.0.x

### Operating system

Linux

### Operating system version

Ubuntu (via github actions docker)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-07-12:
Here is a more simplified demo workflow that reproduces the fault by using Arduino CLI directly rather than depending on the **arduino/compile-sketches** action:

```yaml
on:
  push:

jobs:
  demo:
    runs-on: ubuntu-latest

    steps:
      - uses: actions/checkout@v4
        with:
          repository: arduino/arduino-cli

      - uses: actions/setup-go@v5
        with:
          go-version: 1.21

      - uses: arduino/setup-task@v2
        with:
          repo-token: ${{ secrets.GITHUB_TOKEN }}
          version: 3.x

      - run: |
          task build

      - run: |
          ./arduino-cli core install --additional-urls https://www.pjrc.com/teensy/package_teensy_index.json teensy:avr
          ./arduino-cli sketch new "${{ runner.temp }}/Foo"

      - run: |
          ./arduino-cli compile --fqbn teensy:avr:teensy36 "${{ runner.temp }}/Foo"
```

cmaglie, 2024-07-17:
@tfry-git thanks for the bug report, and thanks @per1234 for the slimmed-down example.

After some testing, I found that **while running under CI** the latest CLI runs the Teensy loader, while the 0.35.x does not.
Instead in a desktop environment, the Teensy loader is always executed.

Investigating...

cmaglie, 2024-07-19:
Fixed by #2669, it will be released on Arduino CLI 1.0.3.

closed 2024-07-19 by cmaglie

cmaglie, 2024-07-23:
Arduino CLI 1.0.3 has been released, the gh action should run without issues now.