arduino/arduino-cli #2661: arduino-cli

atannen, 2024-07-09:
### Describe the problem

This is a problem in arduino-cli_1.0.2 monitor, I did not see it in releases before 1.0.2.

$ arduino-cli_1.0.2 monitor -p /dev/ttyACM0

is generating extra output to the display that was not present in previous versions
(this is upon startup, the "new" output is starting with "Using default monitor" etc.)
see below.

The latest nightly build behaves this way too.

### To reproduce

$ arduino-cli_1.0.2 monitor -p /dev/ttyACM0
Using default monitor configuration for board: adafruit:samd:adafruit_feather_m4_can
Monitor port settings:
  baudrate=9600
  bits=8
  dtr=on
  parity=none
  rts=on
  stop_bits=1

Connecting to /dev/ttyACM0. Press CTRL-C to exit.

### Expected behavior

$ arduino-cli_1.0.1 monitor -p /dev/ttyACM0
Connecting to /dev/ttyACM0. Press CTRL-C to exit.

### Arduino CLI version

arduino-cli_1.0.2

### Operating system

Linux

### Operating system version

6.5.0-42-lowlatency (Ubuntu 22.04.4)

### Additional context

my fqbn is adafruit:samd:adafruit_feather_m4_can
but I assume that is not important.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-07-10:
Hi @atannen. The added output is intended. It is communicating the important information about which port settings are being used. If you don't want that, simply add the `--quiet` flag to the command:

https://arduino.github.io/arduino-cli/dev/commands/arduino-cli_monitor/#options

```text
-q, --quiet                        Run in silent mode, show only monitor input and output.
```

closed 2024-07-10 by per1234

atannen, 2024-07-10:
Thank you, @perl1234.  I saw the change in behavior in this version and I didn't see anything about it in the changelog.