arduino/arduino-cli #2662: Error during build: Platform 'Arduino:avr' not found: platform not installed Try running `arduino-cli core install Arduino:avr` even after installing

krnxmid, 2024-07-10:
### Describe the problem

I am new to arduino.I recently installed the arduino cli and started to compile and upload a sketch but i am getting this message: "`Error during build: Platform 'Arduino:avr' not found: platform not installed`"
Try running "`arduino-cli core install Arduino:avr`"
it tried running "`arduino-cli core install Arduino:avr`" and got the message "`Platform arduino:avr@1.8.6 already installed`"
i am confused someone help

### To reproduce

i used `arduino-cli compile --fqbn Arduino:avr:uno Blink.ino`

### Expected behavior

i should be able to compile and upload my code

### Arduino CLI version

1.0.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-07-10:
Hi @krnxmid. The FQBN is case sensitive. You used the vendor name `Arduino` (note the uppercase "A"), but the correct vendor name is `arduino` (note the lowercase "a").

So the correct FQBN is `arduino:avr:uno`.

If you have any other questions about Arduino CLI usage, you are welcome to request assistance over on Arduino Forum:

https://forum.arduino.cc/c/software/arduino-cli/89

We're always happy to help out over there.

closed 2024-07-10 by per1234

krnxmid, 2024-07-13:
Thanks it worked!!!