arduino/arduino-cli #2665: Burn-bootloader does not work with RP2040 boards added

dtomcat, 2024-07-14:
### Describe the problem

Trying to burn bootloader to atmega32u4; but fails because trying to kill process discovery = rp2040.  when I remove that board type, It functions perfectly.

### To reproduce

install Rapsberry pi pico/RP2040 board types by "Earle F. Philhower, III" and try to burn a bootloader.

### Expected behavior

Should burn the bootloader to the atmega32U4

### Arduino CLI version

1.0.2

### Operating system

Windows

### Operating system version

Latest updates

### Additional context

INFO[0007] Executing `arduino-cli burn-bootloader`
ERRO[0008] Killing discovery process: TerminateProcess: Access is denied.  discovery=rp2040
ERRO[0008] Waiting discovery process termination: exit status 0xc0000409  discovery=rp2040
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x8 pc=0xbe7f5e]

goroutine 44 [running]:
github.com/arduino/pluggable-discovery-protocol-handler/v2.(*Client).Run(0xc0001f8870)
        /go/pkg/mod/github.com/arduino/pluggable-discovery-protocol-handler/v2@v2.2.0/client.go:281 +0x19e
github.com/arduino/arduino-cli/internal/arduino/discovery/discoverymanager.(*DiscoveryManager).startDiscovery(0xc000244050, 0xc0001f8870)
        /home/build/internal/arduino/discovery/discoverymanager/discoverymanager.go:203 +0x95
github.com/arduino/arduino-cli/internal/arduino/discovery/discoverymanager.(*DiscoveryManager).Start.func2(0x736169626f742f6d?)
        /home/build/internal/arduino/discovery/discoverymanager/discoverymanager.go:105 +0x3c
created by github.com/arduino/arduino-cli/internal/arduino/discovery/discoverymanager.(*DiscoveryManager).Start in goroutine 1
        /home/build/internal/arduino/discovery/discoverymanager/discoverymanager.go:104 +0x1ae

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2024-09-13:
I think I've identified the issue, I'm preparing a fix for this problem.

cmaglie, 2024-09-13:
@dtomcat 
if you can still reproduce the issue, may you test the `arduino-cli` build here (scroll down at the bottom of the page): https://github.com/arduino/arduino-cli/actions/runs/10849485867?pr=2707

It contains a fix for the pluggable-discovery client.

dtomcat, 2024-09-13:
I’ve disassembled the setup and have a few projects… I’m not sure when I’ll have time… but can definitely give it a go when I do have the time if someone else hasn’t done so. Thank you for the fix!

PR #2707 (2024-10-07): Updated pluggable-discovery-protocol-handler library
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [X] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / featur [... cut]
closed 2024-10-07 by cmaglie