arduino/arduino-cli #2667: Failed to install using bash on Windows

lucasssvaz, 2024-07-16:
### Describe the problem

Since the latest release I can't install through the script. It originally started happening in our CI and I tested locally to make sure it is also happening.

https://github.com/espressif/arduino-esp32/actions/runs/9957218514/job/27508776142

As seen on the CI, this only happens in Windows.

### To reproduce

```
lucas@Desktop-Win MINGW64 ~
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
Installing in /c/Users/lucas/bin
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.2_Windows_64bit.zip
Failed to install arduino-cli

lucas@Desktop-Win MINGW64 ~
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s 1.0.1
Installing in /c/Users/lucas/bin
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.1_Windows_64bit.zip
Failed to install arduino-cli

lucas@Desktop-Win MINGW64 ~
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s nightly-latest
Installing in /c/Users/lucas/bin
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/nightly/arduino-cli_nightly-latest_Windows_64bit.zip
Failed to install arduino-cli

lucas@Desktop-Win MINGW64 ~
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=. sh
Installing in .
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.2_Windows_64bit.zip
Failed to install arduino-cli
```

### Expected behavior

Normal installation.

### Arduino CLI version

1.0.2

### Operating system

Windows

### Operating system version

Windows 10/`windows-latest`

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

pillo79, 2024-07-24:
cc @umbynos

umbynos, 2024-07-25:
Hello, I just tried on a Windows machine with [git bash](https://gitforwindows.org/) installed and everything seems to work just fine:
```
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
Installing in /c/Users/Arduino/bin
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip
Archive:  /tmp/arduino-cli_1.0.3_Windows_64bit.zip
  inflating: /tmp/arduino-cli/arduino-cli.exe
  inflating: /tmp/arduino-cli/LICENSE.txt
arduino-cli  Version: 1.0.3 Commit: 8b6ad258 Date: 2024-07-23T08:45:25Z installed successfully in /c/Users/Arduino/bin
```

lucasssvaz, 2024-07-25:
Just run it again with git bash on WIn 10 and same error happens:

```
lucas@Desktop-Win MINGW64 ~
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh
Installing in /c/Users/lucas/bin
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip
Failed to install arduino-cli
```

cmaglie, 2024-07-25:
I tried the same and on my PC I get:
```
+++ curl -s -w '%{http_code}' -L https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip -o /tmp/arduino-cli_1.0.3_Windows_64bit.zip
++ GETFILE_HTTP_STATUS_CODE=200
++ echo 200
+ httpStatusCode=200
```

it seems that your `curl` returns `000` instead of `200` 🤔 
could you try:
```
curl -s -w '%{http_code}' -L https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip -o /tmp/arduino-cli_1.0.3_Windows_64bit.zip
```

cmaglie, 2024-07-25:
Also a `curl --version` mine is:

```
$ curl --version
curl 7.61.1 (x86_64-w64-mingw32) libcurl/7.61.1 OpenSSL/1.0.2p (WinSSL) zlib/1.2.11 libidn2/2.0.5 nghttp2/1.33.0
Release-Date: 2018-09-05
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtsp smtp smtps telnet tftp
Features: AsynchDNS IDN IPv6 Largefile SSPI Kerberos SPNEGO NTLM SSL libz TLS-SRP HTTP2 HTTPS-proxy MultiSSL Metalink
```

lucasssvaz, 2024-07-25:
```
lucas@Desktop-Win MINGW64 ~
$ curl -s -w '%{http_code}' -L https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip -o /tmp/arduino-cli_1.0.3_Windows_64bit.zip
000

lucas@Desktop-Win MINGW64 ~
$ curl --version
curl 8.8.0 (x86_64-w64-mingw32) libcurl/8.8.0 Schannel zlib/1.3.1 brotli/1.1.0 zstd/1.5.6 libidn2/2.3.7 libpsl/0.21.5 libssh2/1.11.0
Release-Date: 2024-05-22
Protocols: dict file ftp ftps gopher gophers http https imap imaps ipfs ipns ldap ldaps mqtt pop3 pop3s rtsp scp sftp smb smbs smtp smtps telnet tftp
Features: alt-svc AsynchDNS brotli HSTS HTTPS-proxy IDN IPv6 Kerberos Largefile libz NTLM PSL SPNEGO SSL SSPI threadsafe UnixSockets zstd
```

cmaglie, 2024-07-25:
Please, try:
```
curl -I -L https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip
```
(sorry for the long testing I'm a bit lost here... I'm trying to figure out what's happening with your `curl`)

lucasssvaz, 2024-07-25:
```
$ curl -I -L https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 15.2M    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0HTTP/1.1 200 OK
Date: Thu, 25 Jul 2024 19:33:01 GMT
Content-Type: application/zip
Content-Length: 15942283
Connection: keep-alive
CF-Ray: 8a8e94497e5f82e5-GIG
CF-Cache-Status: HIT
Accept-Ranges: bytes
Age: 110287
ETag: "438dd459e3cc8f312ef882d68fe80c7e-2"
Last-Modified: Tue, 23 Jul 2024 08:50:16 GMT
Strict-Transport-Security: max-age=31536000; includeSubDomains; preload
Vary: Accept-Encoding
Via: 1.1 50fb19eda678e6a896981a444fb09aa6.cloudfront.net (CloudFront)
x-amz-cf-id: EdXDluf2OaIATJSTfNQyCMWwwUWccwY9vL_N2Z0sHOL3-zfg9r-a9w==
x-amz-cf-pop: MIA3-P3
x-amz-id-2: UuCPkJ05XTtqAYF6C/j1UuPYSF4VkHTdHn0n7Ncf7sXObkC4kqyFPB+H47ZSpvK/cjh8uUw43gWmWkaD7F/uFA==
x-amz-replication-status: COMPLETED
x-amz-request-id: FS133ZNW0HKYM6RF
x-amz-server-side-encryption: AES256
x-amz-version-id: ZCRd.9ppRtjB1b51.DEElxqFRdJLGARl
x-cache: Miss from cloudfront
Server: cloudflare
```

lucasssvaz, 2024-07-26:
If I try using `latest` rather than a specific version, It is downloaded successfully:

```
lucas@Desktop-Win MINGW64 ~
$ curl -I -L https://downloads.arduino.cc/arduino-cli/arduino-cli_latest_Windows_64bit.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0 15.2M    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0HTTP/1.1 302 Found
Date: Fri, 26 Jul 2024 12:02:46 GMT
Content-Type: application/json
Content-Length: 0
Connection: keep-alive
Location: https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip
CF-Ray: 8a943e1a48f182f1-GIG
CF-Cache-Status: HIT
Age: 88
Strict-Transport-Security: max-age=31536000; includeSubDomains; preload
Vary: Accept-Encoding
Via: 1.1 5a60386eb046f88a482c3726fc19fb4e.cloudfront.net (CloudFront), 1.1 e759cef9ef04dc6632a71818dfac3a76.cloudfront.net (CloudFront)
x-amz-apigw-id: bhNyMG1_oAMEG5Q=
x-amz-cf-id: lFbyX5qRthTRActS5gd7hyhfXyM_f7ToYOdJ_NEc2HE1cDk24eXRFg==
x-amz-cf-pop: MIA3-P2
x-amz-cf-pop: MIA3-P3
x-amzn-requestid: b1d89b4a-6cf4-4f48-af9e-89718a7a066e
x-amzn-trace-id: Root=1-66a3900d-085ddee755022f34504bfb87;Parent=5c0d5af354d1b789;Sampled=0;lineage=5405e195:0
x-cache: Miss from cloudfront
Server: cloudflare

HTTP/1.1 200 OK
Date: Fri, 26 Jul 2024 12:02:46 GMT
Content-Type: application/zip
Content-Length: 15942283
Connection: keep-alive
CF-Ray: 8a943e1dce2382f1-GIG
CF-Cache-Status: HIT
Accept-Ranges: bytes
Age: 169672
ETag: "438dd459e3cc8f312ef882d68fe80c7e-2"
Last-Modified: Tue, 23 Jul 2024 08:50:16 GMT
Strict-Transport-Security: max-age=31536000; includeSubDomains; preload
Vary: Accept-Encoding
Via: 1.1 50fb19eda678e6a896981a444fb09aa6.cloudfront.net (CloudFront)
x-amz-cf-id: EdXDluf2OaIATJSTfNQyCMWwwUWccwY9vL_N2Z0sHOL3-zfg9r-a9w==
x-amz-cf-pop: MIA3-P3
x-amz-id-2: UuCPkJ05XTtqAYF6C/j1UuPYSF4VkHTdHn0n7Ncf7sXObkC4kqyFPB+H47ZSpvK/cjh8uUw43gWmWkaD7F/uFA==
x-amz-replication-status: COMPLETED
x-amz-request-id: FS133ZNW0HKYM6RF
x-amz-server-side-encryption: AES256
x-amz-version-id: ZCRd.9ppRtjB1b51.DEElxqFRdJLGARl
x-cache: Miss from cloudfront
Server: cloudflare
```

lucasssvaz, 2024-07-26:
```
lucas@Desktop-Win MINGW64 ~
$ curl -s -w '%{http_code}\n' -fL https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip ; echo "Exit code: $?"
000
Exit code: 43
```

From MAN:

```
43
Internal error. A function was called with a bad parameter.
```

BTW, If I do:
```
curl -L https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.3_Windows_64bit.zip -o arduino-cli.zip
```
It downloads fine.

cmaglie, 2024-07-30:
At this point, I'm wondering if it's better to check both the exit code AND the HTTP status code (that may be `000` that means... "something" happened...).
I'm also keen to remove with `-s` doesn't seem like a good option

cmaglie, 2024-10-25:
@lucasssvaz 
sorry for the long delay.
I've made a patch to `install.sh`, now it should display a better error message thanks to a `--show-error` flag:

```
$ curl -s http://asdhkjashdkjahd.com                            <- no error displayed due to -s
$ curl -s http://asdhkjashdkjahd.com --show-error
curl: (6) Could not resolve host: asdhkjashdkjahd.com
```

If you're still experiencing the error, would you like to give it a try?

The patch is here https://github.com/arduino/arduino-cli/pull/2738
The updated install.sh can be downloaded here: https://raw.githubusercontent.com/arduino/arduino-cli/aed5066b2e52091aea87f5e992eb280be38c81d5/install.sh

lucasssvaz, 2024-10-25:
@cmaglie Just tested, still some issue with `libcurl`:

```
lucas@Desktop-Win MINGW64 ~
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/aed5066b2e52091aea87f5e992eb280be38c81d5/install.sh | sh
Installing in /c/Users/lucas/bin
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.4_Windows_64bit.zip
curl: (43) A libcurl function was given a bad argument
Failed to install arduino-cli
```

cmaglie, 2024-10-25:
It seems to be a problem with curl 8.8.0 -> https://github.com/curl/curl/issues/13845
Can you try to update `curl`?

lucasssvaz, 2024-10-25:
I am using the curl version packed with git and even trying to update it fails, so probably this is the reason XD

```
C:\Users\lucas>git update-git-for-windows
Git for Windows 2.45.2.windows.1 (64-bit)
Update 2.47.0.windows.2 is available
curl: (43) A libcurl function was given a bad argument
```

I will reinstall everything and report back

lucasssvaz, 2024-10-25:
Now it works as expected, it was an issue with the curl version packed with git for windows. Thanks for the help debugging!

```
lucas@Desktop-Win MINGW64 ~
$ curl --version
curl 8.10.1 (x86_64-w64-mingw32) libcurl/8.10.1 Schannel zlib/1.3.1 brotli/1.1.0 zstd/1.5.6 libidn2/2.3.7 libpsl/0.21.5 libssh2/1.11.0
Release-Date: 2024-09-18
Protocols: dict file ftp ftps gopher gophers http https imap imaps ipfs ipns ldap ldaps mqtt pop3 pop3s rtsp scp sftp smb smbs smtp smtps telnet tftp
Features: alt-svc AsynchDNS brotli HSTS HTTPS-proxy IDN IPv6 Kerberos Largefile libz NTLM PSL SPNEGO SSL SSPI threadsafe UnixSockets zstd

lucas@Desktop-Win MINGW64 ~
$ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/aed5066b2e52091aea87f5e992eb280be38c81d5/install.sh | sh
Installing in /c/Users/lucas/bin
ARCH=64bit
OS=Windows
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_1.0.4_Windows_64bit.zip
Archive:  /tmp/arduino-cli_1.0.4_Windows_64bit.zip
  inflating: /tmp/arduino-cli/arduino-cli.exe
  inflating: /tmp/arduino-cli/LICENSE.txt
arduino-cli  Version: 1.0.4 Commit: a0d912da Date: 2024-08-12T13:42:36Z installed successfully in /c/Users/lucas/bin
```

cmaglie, 2024-10-25:
Thank you for your responsiveness!

Even if not related to Arduino CLI, at least now:
* we have improved the error reporting
* we now know where this `curl: (43) A libcurl function was given a bad argument` comes from!

closed 2024-10-25 by cmaglie