arduino/arduino-cli #2671: Cyrillic characters in the path lead to all code recompiling every time

vvb333007, 2024-07-20:
### Describe the problem

I am using Windows10, and my user name is in Cyrillic letters.  
[image]
Every sketch and every library gets recompiled every time I press "Compile button" even if no files were changed. This happens because .d (dependencies) files contain wrong path.  Example below shows the problem win RTCLib but you can use any library/code to reproduce

I checked .d files in a temp folder and this is what I found inside:

(must be "C:\Users\ֲВячеслав....")

[code]
C:\Users\ֲקוסכאג\AppData\Local\Temp\arduino\sketches\5E52C855D96EEA0FECA98993020C8EE1\libraries\RTClib\RTC_DS1307.cpp.o: \
 c:\Users\ֲקוסכאג\Desktop\Arduino\libraries\RTClib\src\RTC_DS1307.cpp \
 c:\Users\ֲקוסכאג\Desktop\Arduino\libraries\RTClib\src\RTClib.h \
 c:\Users\ֲקוסכאג\Desktop\Arduino\libraries\Adafruit_BusIO/Adafruit_I2CDevice.h \
[/code]

So no wonder that is "up-to-date" check fails leading to full recompiling. Some paths are Ok tho.  Looks like the bug in Arduino-IDE or/and tools it uses.

PS: All the .o and .a files do exist in temp folder but get recompiled every time

### To reproduce

1. Use cyrillic characters in Windows10 user name (my sketchfolder is located on a Desktop, so sketchfolder includes these cyrillic characters) 
2. Place your sketch folder on a desktop (mine is located at "C:\Users\Вячеслав\Desktop\Arduino" )
3. Load any sketch example (RTCLib examples, TinyGSM) OR your own sketch which uses SPI library to see how SPI lib gets recompiled
4. 
5. Press compile 
6. Press compile
7. ...

### Expected behavior

Expected behaviour:  not to recompile files which were not modified

### Arduino IDE version

2.3.2, CLI version 0.35.3

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2024-10-15:
@vvb333007 I've implemented a tentative workaround here:
https://github.com/arduino/arduino-cli/pull/2733

could you try a test build? (you can find at [the bottom of this page](https://github.com/arduino/arduino-cli/actions/runs/11344942736))

vvb333007, 2024-10-15:
> [@vvb333007](https://github.com/vvb333007) I've implemented a tentative workaround here: [#2733](https://github.com/arduino/arduino-cli/pull/2733)
> 
> could you try a test build? (you can find at [the bottom of this page](https://github.com/arduino/arduino-cli/actions/runs/11344942736))

Yes but it will take some time as I never used arduino-cli. What version (and where?) should I download?

vvb333007, 2024-10-15:
> [@vvb333007](https://github.com/vvb333007) I've implemented a tentative workaround here: [#2733](https://github.com/arduino/arduino-cli/pull/2733)
> 
> could you try a test build? (you can find at [the bottom of this page](https://github.com/arduino/arduino-cli/actions/runs/11344942736))

Well it still recompiles everything. 

And .d files have cyrillic symbols in some unusual encoding (i.e.  not totally damaged as when it is simply replaced with "?").

[image]

cmaglie, 2024-10-16:
Thanks for testing it!
Could you paste the output of the compile with the `--log --log debug` flags added?

cmaglie, 2024-10-16:
> And .d files have cyrillic symbols in some unusual encoding (i.e. not totally damaged as when it is simply replaced with "?").

Those .d files are directly created by g++/gcc, but for some reason, they are not UTF-8 encoded.
Looking at the documentation of the Win32 API, it's not clear what is [the default encoding](https://learn.microsoft.com/en-us/windows/win32/intl/code-pages), instead it seems dependent on which part of the Win32 API is used in the application. I suspect that gcc/g++ uses an old API that writes using "codepages" which would make decoding those files a nightmare because the codepage depends on the currently selected locale.

To help me, I'd like to ask you:
* The currently selected language on your PC
* If you can provide a copy of a .d file together and the correct spelling of the path written in it (so I can try to figure out the encoding).

Thanks!

cmaglie, 2024-10-16:
Ok, after some research, I may have found the key to solving this issue.

Please test this build before, if it works I don't need anything else: https://github.com/arduino/arduino-cli/actions/runs/11362531613

vvb333007, 2024-10-16:
> Ok, after some research, I may have found the key to solving this issue.
> 
> Please test this build before, if it works I don't need anything else: https://github.com/arduino/arduino-cli/actions/runs/11362531613

Yay! It works. .d files now contain valid cyrillic paths. It reports that it uses precompiled .o files. Great!

vvb333007, 2024-10-16:
There are no ???? strings anymore. Examined whole output - it is all clean in proper encoding

cmaglie, 2024-10-16:
That's fantastic news @vvb333007, thanks for your help in debugging this issue!

vvb333007, 2024-10-16:
> Thanks for testing it! Could you paste the output of the compile with the `--log --log-level debug` flags added?
[code]
[36mINFO[0m[0000] arduino-cli version test-2733-e07b5117a6adc575eb7cd0c6cbabdac5a96e42c9-git-snapshot 
[36mINFO[0m[0000] Using config file: C:\Users\Вячеслав\AppData\Local\Arduino15\arduino-cli.yaml 
[36mINFO[0m[0000] Executing `arduino-cli compile`              
[36mINFO[0m[0000] Reading sketch from dir: E:\Arduino\espshell 
[33mWARN[0m[0000] Error checking for updates: no major version found 
[36mINFO[0m[0000] Checking signature                            [36mindex[0m="C:\\Users\\Вячеслав\\AppData\\Local\\Arduino15\\package_index.json" [36msignatureFile[0m="C:\\Users\\Вячеслав\\AppData\\Local\\Arduino15\\package_index.json.sig" [36mtrusted[0m=true
[36mINFO[0m[0000] Loading hardware from: C:\Users\Вячеслав\AppData\Local\Arduino15\packages 
[36mINFO[0m[0000] Loading package arduino from: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\arduino\hardware 
[36mINFO[0m[0000] Checking existence of 'tools' path: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\arduino\tools 
[36mINFO[0m[0000] Loading tools from dir: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\arduino\tools 
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="arduino:dfu-util@0.11.0-arduino5"
[36mINFO[0m[0000] Loading package builtin from: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\builtin 
[36mINFO[0m[0000] Checking existence of 'tools' path: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\builtin\tools 
[36mINFO[0m[0000] Loading tools from dir: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\builtin\tools 
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="builtin:ctags@5.8-arduino11"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="builtin:dfu-discovery@0.1.2"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="builtin:mdns-discovery@1.0.9"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="builtin:serial-discovery@1.4.1"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="builtin:serial-monitor@0.14.1"
[36mINFO[0m[0000] Loading package esp32 from: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\esp32\hardware 
[36mINFO[0m[0000] Missing signature file                        [36mindex[0m="C:\\Users\\Вячеслав\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5\\installed.json"
[36mINFO[0m[0000] Adding monitor tool                           [36mprotocol[0m=serial [36mtool[0m="builtin:serial-monitor"
[36mINFO[0m[0000] Loaded platform                               [36mplatform[0m="esp32:esp32@3.0.5"
[36mINFO[0m[0000] Checking existence of 'tools' path: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\esp32\tools 
[36mINFO[0m[0000] Loading tools from dir: C:\Users\Вячеслав\AppData\Local\Arduino15\packages\esp32\tools 
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:esp-rv32@2302"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:esp-x32@2302"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:esp-xs2@2302"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:esp-xs3@2302"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:esp32-arduino-libs@idf-release_v5.1-33fbade6"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:esptool_py@4.6"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:mklittlefs@3.0.0-gnu12-dc7f933"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:mkspiffs@0.2.3"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:openocd-esp32@v0.12.0-esp32-20240821"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:riscv32-esp-elf-gdb@12.1_20231023"
[36mINFO[0m[0000] Loaded tool                                   [36mtool[0m="esp32:xtensa-esp-elf-gdb@12.1_20231023"
[37mDEBU[0m[0000] current best: esp32:esp32@3.0.5               [37mmanaged[0m=true [37mversion[0m=3.0.5
[36mINFO[0m[0000] Adding libraries dir                          [36mdir[0m="C:\\Users\\Вячеслав\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.5\\libraries" [36misSingleLibrary[0m=false [36mlocation[0m=platform
[36mINFO[0m[0000] Loading libraries index file                  [36mindex[0m="C:\\Users\\Вячеслав\\AppData\\Local\\Arduino15\\library_index.json"
[36mINFO[0m[0000] Adding libraries dir                          [36mdir[0m="C:\\Users\\Вячеслав\\Documents\\Arduino\\libraries" [36misSingleLibrary[0m=false [36mlocation[0m=user
[37mDEBU[0m[0001] current best: esp32:esp32@3.0.5               [37mmanaged[0m=true [37mversion[0m=3.0.5
[37mDEBU[0m[0001] Required tool                                 [37mtool[0m="arduino:dfu-util@0.11.0-arduino5"
[37mDEBU[0m[0001] Required tool                                 [37mtool[0m="esp32:esp-rv32@2302"
[37mDEBU[0m[0001] Required tool                                 [37mtool[0m="esp32:esp-x32@2302"
[37mDEBU[0m[0001] Required tool                                 [37mtool[0m="esp32:esp-xs2@2302"
[... cut ...]
"C:\\Users\\Вячеслав\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2302/bin/xtensa-esp32-elf-size" -A "C:\\Users\\Вячеслав\\AppData\\Local\\arduino\\sketches\\C7FAA3C305436E6DB4A2D213C3851A1B/espshell.ino.elf"
Скетч использует 377173 байт (28%) памяти устройства. Всего доступно 1310720 байт.
Глобальные переменные используют 22400 байт (6%) динамической памяти, оставляя 305280 байт для локальных переменных. Максимум: 327680 байт.

[92mUsed platform[0m [92mVersion[0m [90mPath[0m
[93mesp32:esp32[0m   3.0.5   [90mC:\Users\Вячеслав\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5[0m

[/code]

vvb333007, 2024-10-16:
The only problem with encoding is this lines:

esptool.py v4.6
Wrote 0x400000 bytes to file C:\Users\��������\AppData\Local\arduino\sketches\C7FAA3C305436E6DB4A2D213C3851A1B/espshell.ino.merged.bin, ready to flash to offset 0x0

But this does not affect compiling or/and uploading.

JFY: lines "Используем предварительно скомпилированный файл:" in the output in the previous comment mean "Using precompiled file". There are both english and russian messages about using precompiled files

closed 2024-10-16 by cmaglie