arduino/arduino-cli #2683: Code no longer compiles when using `--build-property "compiler.cpp.extra_flags=-DPRINT_DEPENDENCIES"`

tyeth, 2024-08-01:
### Describe the problem

As advised in the help page for compile (rubbish docs as there are just 3 examples and no links to available build properties or flag options)

### To reproduce

Build Adafruit wippersnapper arduino firmware using arduino-cli and --verbose for the adafruit metro esp32s2 (works)
Do the same then with --verbose --build-property "compiler.cpp.extra_flags=-DPRINT_DEPENDENCIES" and you'll see the new define is included but the cpp flags file no longer is loaded with a -c argument beforehand (where the -DPrint_Deps appears).
Compile no longer works.
Adding the -c to the extra flags makes compile work again:
`--build-property "compiler.cpp.extra_flags=-DPRINT_DEPENDENCIES -c"`

### Expected behavior

Adding an extra build flag using the --build-property build.extra_flags would not remove default flags (-DESP32 disappears), so I'm instead using compiler.cpp.extra_flags but that breaks the compile.

### Arduino CLI version

1.0.3

### Operating system

Windows

### Operating system version

11

### Additional context

ESP32 BSP 3.0.3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-08-01:
Thanks for taking the time to submit an issue @tyeth.

As you discovered, when you set the value of a property via the `--build-property` flag, it overrides whatever value might have been set in the platform configuration files. If you look in the platform configuration files, you will find this:

https://github.com/espressif/arduino-esp32/blob/3.0.3/platform.txt#L69

```text
compiler.cpp.extra_flags=-MMD -c
```

Even though the idea behind the `compiler.cpp.extra_flags` property was to provided a dedicated property for sole use by the user to inject arbitrary flags into the C++ compilation command, the powerful and flexible Arduino boards platform framework doesn't impose any restrictions on platform developers. Unfortunately some platform developers have pointlessly (because they could instead have simply defined an arbitrary property for the purpose) made use of this property internally.

So the results you experienced are expected when using `--build-property` to override the "esp32" boards platform's `compiler.cpp.extra_flags` property. Arduino CLI is acting exactly as intended so I will close this as off topic. You can report the inappropriate internal usage of these properties by the "esp32" boards platform to the repository where that codebase is hosted:

https://github.com/espressif/arduino-esp32

The `--build-property` flag is a very powerful feature of Arduino CLI that is only intended to be used by very advanced users who have a full understanding of the internals of their board's platform. So it is essential for you to carefully study the platform configuration files to understand the impact of overriding a property if you are going to use this feature.

We are already tracking the need for Arduino to either clearly document the intended purpose of these properties, or perhaps to give them up as a lost cause and define a fresh set of such properties in order to reduce the chances of this type of problem at https://github.com/arduino/arduino-cli/issues/846.

closed 2024-08-01 by per1234

tyeth, 2024-08-01:
Thanks @per1234 
What's the correct way to add an extra define via command line arguments then, and not removing/altering existing defines?

I'm doing this for multiple build targets in CI so just want to add one define for all targets in a build matrix to the building of  some examples from the wippersnapper repo.

per1234, 2024-08-02:
> What's the correct way to add an extra define via command line arguments then, and not removing/altering existing defines?

Just as you are already doing. Just make sure that if the platform author already put content in the property you are overriding, that you include that content in your override (unless you intentionally want to remove that content):

```text
--build-property "compiler.cpp.extra_flags=-DPRINT_DEPENDENCIES -MMD -c"
```