arduino/arduino-cli #2716: ESP32 Nano core 2.0.18 does not compile due to long core folder name 2.0.18-20240930.arduino3

ednieuw, 2024-10-03:
### Describe the problem

If core 2.0.18 in installed sketches fail to compile in IDE2 but will compile in IDE1

IDE2 can not cope with the too long folder name of 2.0.18

### To reproduce

Just compile a sketch --> gives errors

In this folder
c:\Users\abcde\AppData\Local\Arduino15\packages\arduino\hardware\esp32\

rename the folder
2.0.18-20240930.arduino3

to
2.0.18

### Expected behavior

No errors

### Arduino IDE version

2.3.3

### Operating system

Windows

### Operating system version

W11

### Additional context

I can not find the location in Github of the Arduino Nano ESP32 core that causes this error

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-10-03:
Hi @ednieuw.

> Just compile a sketch --> gives errors

It doesn't give errors for me, so we're going to need you to provide more information. Please do this:

1. Change the folder name back to the original `2.0.18-20240930.arduino3`
1. Select **File > Preferences...** from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Check the box next to "**Show verbose output during: > ☐ compile**" in the "**Preferences**" dialog.
1. Click the "**OK**" button.
   The "**Preferences**" dialog will close.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to fail.
1. You will see a "**Compilation error: ...**" notification at the bottom right corner of the Arduino IDE window. Click the "**COPY ERROR MESSAGES**" button on that notification.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

sterretjeToo, 2024-10-03:
Just adding to @per1234's reply

If you want to keep your user name anonymous, please replace it by something with the same length and the same number of spaces, same odd characters (if any) and so on. As mentioned on the forum, I compiled your sketch without issues and by coincedence my user name is 5 characters (just like your abcde).

sterretjeToo, 2024-10-03:
I did look at your zip. 

It goes wrong at line 71 which has a length of 32531 characters; this is very close to 32k which might (or might not) be a coincidence. All other lines up to that point for "Resolve library" are shorter.

Even if you rename the package directory to a shorter name as you did, I'm reasonably sure that you will run into the same problem if your code requires more libraries.

ednieuw, 2024-10-04:
I changed the folder name to 2.0.18-rc3 as suggested by the Arduino team. This resolved the issue for me for now.

martin-eden, 2024-10-21:
Joys of Windows (tm) development.

Problem is that command line length for `cmd.exe` is limited to 32KiB. I believe it's from 80386 times and `command.com`.

Moving directories to root level and renaming them to one-character names would help. But the main problem is hardcoded 32KiB limit on your operation system.

Under GNU/Linux systems command line length is configurable parameter. For my case it is 2MiB.

```
$ getconf ARG_MAX
2097152
```

sterretjeToo, 2024-11-06:
Based on the error log in https://github.com/arduino/arduino-cli/issues/2716#issuecomment-2391752801 I've tried to recreate the errors. One can make the compile process fail based on below description.

OS: Win11
IDE 2.3.3
Board package
FQBN: arduino:esp32:nano_nora
Using board 'nano_nora' from platform in folder: C:\Users\bugge\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.18-20240930.arduino3
Using core 'esp32' from platform in folder: C:\Users\bugge\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.18-20240930.arduino3

```cpp
//#include <Preferences.h>
//#include <Adafruit_NeoPixel.h>
#include <NimBLEDevice.h>  // library
//#include <Ticker.h>
#include <ESPAsyncWebServer.h> // library
//#include <AsyncTCP.h>  // library
//#include <FS.h>
//#include <WiFi.h>

void setup()
{
  // put your setup code here, to run once:
}

void loop()
{
  // put your main code here, to run repeatedly:
}
```
This sketch does compile.

Next I went in steps to enabling some more libraries.
```cpp
//#include <Preferences.h>
#include <Adafruit_NeoPixel.h>
#include <NimBLEDevice.h>  // library
#include <Ticker.h>
#include <ESPAsyncWebServer.h> // library
//#include <AsyncTCP.h>  // library
//#include <FS.h>
//#include <WiFi.h>
```
It still compiled.

In the last step, enabling Preferences.h, it failed.

```cpp
#include <Preferences.h>
#include <Adafruit_NeoPixel.h>
#include <NimBLEDevice.h>  // library
#include <Ticker.h>
#include <ESPAsyncWebServer.h> // library
//#include <AsyncTCP.h>  // library
//#include <FS.h>
//#include <WiFi.h>
```

Full output attached (err01.txt); check the output for CreateProcess error. Note that something like below is missing in favour ( 🙂 ) of a ping timeout error popup
```text
exit status 1

Compilation error: exit status 1
```
[err01.txt](https://github.com/user-attachments/files/17648819/err01.txt)

---
There are a few variations possible that all result in errors. After above I disabled the NimBLEDevice library and the compile still fails on the Preferences library
```cpp
#include <Preferences.h>
#include <Adafruit_NeoPixel.h>
//#include <NimBLEDevice.h>  // library
#include <Ticker.h>
#include <ESPAsyncWebServer.h> // library
//#include <AsyncTCP.h>  // library
//#include <FS.h>
//#include <WiFi.h>
```

Full output attached (err02.txt)

[err02.txt](https://github.com/user-attachments/files/17648830/err02.txt)

Note that we now get the `Compilation error: exit status 1` at the end and I did not get a ping timeout error.

---
A number of variations are possible with the includes. The number of CreateProcess errors seems to have a relation with the number of source files in the affected library after clearing the cache (both cores and sketches).

sterretjeToo, 2024-11-15:
Another one to add to the list: https://forum.arduino.cc/t/createprocess-error-when-uploading-esp32-iot-cloud-code-from-arduino-ide/1322028/

Issue confirmed with the code presented in the topic; log attached.

[error.txt](https://github.com/user-attachments/files/17770483/error.txt)

Just an observation:
After changing the package name it compiles. After changing it back it still compiles (till such time that the cache is cleared).

pillo79, 2024-11-21:
Core `2.0.18-arduino.5` has just been published and should _practically solve_ this issue.

---
 
There are 2 fundamental Windows limitations derived from decades of legacy:
 
1. any **command lines** must be [smaller than ~32kB](https://devblogs.microsoft.com/oldnewthing/20031210-00/?p=41553);
2. each **file path** must be [at most 260 chars](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry), unless the requesting app is specially crafted.

Point 1 is solved by converting, in each compiler call, ~200 instances of the core install path to a single one which is referenced multiple times. This reduced in my case a ~28kB command line into a ~10k one - way below the limit, and the multiplier for extra chars in the user name is now 3 and not 200! 🎉.

Point 2 is only mitigated, since most of the install path is not directly under control. However, the longest path used by include files has been shortened by 30 chars by renaming the accessible "tool name" and "tool version" components. Compared with the much smaller core version name, this should solve the issue for anybody with a user name shorter than ~50 chars, if my maths is correct.

pillo79, 2024-11-25:
Closing as confirmed by OP here: https://forum.arduino.cc/t/core-2-0-18-gives-compile-errors-in-ide2-and-how-to-solve/1307544