arduino/arduino-cli #2719: `cc.arduino.cli.commands.v1.ArduinoCoreService.Init` hangs if `Monitor` is running

per1234, 2024-10-06:
### Describe the problem

A client might make additional [`cc.arduino.cli.commands.v1.ArduinoCoreService.Init`](https://arduino.github.io/arduino-cli/dev/rpc/commands/#initrequest) requests.

For example, Arduino IDE periodically (and on demand) performs the following sequence:

1. [Update package and/or library index](https://github.com/arduino/arduino-ide/blob/77136687d374dce1660191059427f1941f4cef16/arduino-ide-extension/src/node/core-client-provider.ts#L280-L285)
1. [Initialize gRPC client instance](https://github.com/arduino/arduino-ide/blob/77136687d374dce1660191059427f1941f4cef16/arduino-ide-extension/src/node/core-client-provider.ts#L297)

🐛 If there is a running [`cc.arduino.cli.commands.v1.ArduinoCoreService.Monitor`](https://arduino.github.io/arduino-cli/dev/rpc/commands/#ccarduinoclicommandsv1monitorproto) process, the `Init` request hangs.

### To reproduce

#### Equipment

- Any board or other device that will produce a serial port on your computer.

#### Demo

1. Open a terminal.
1. Start the Arduino CLI gRPC daemon:
   ```text
   $ arduino-cli version

   arduino-cli  Version: git-snapshot Commit: 863c1ec36 Date: 2024-10-06T06:54:36Z

   $ arduino-cli daemon

   Daemon is now listening on 127.0.0.1:50051
   {"IP":"127.0.0.1","Port":"50051"}
   ```
1. Open another terminal.
1. Run the following [`grpcurl`](https://github.com/fullstorydev/grpcurl) command in the new terminal to create a gRPC client instance:
   ```text
   $ grpcurl \
     -plaintext \
     -import-path ./rpc \
     -proto cc/arduino/cli/commands/v1/commands.proto \
     127.0.0.1:50051 \
     cc.arduino.cli.commands.v1.ArduinoCoreService.Create
   
   {
     "instance": {
       "id": 1
     }
   }
   ```
1. Run the following command to initialize the instance:
   ```text
   $ grpcurl \
     -plaintext \
     -import-path ./rpc \
     -proto cc/arduino/cli/commands/v1/commands.proto \
     -d '{"instance": {"id": 1}}' \
     127.0.0.1:50051 \
     cc.arduino.cli.commands.v1.ArduinoCoreService.Init
   ```
1. Run the following command to start a Monitor request:
   ```text
   $ grpcurl \
     -plaintext \
     -import-path ./rpc \
     -proto cc/arduino/cli/commands/v1/commands.proto \
     -d @ \
     127.0.0.1:50051 \
     cc.arduino.cli.commands.v1.ArduinoCoreService.Monitor
   ```
1. Send the request data to open the available serial port:
   ❗ Adjust the value of `open_request.port.address` as is appropriate for the port on your computer.
   ```json
   {
     "open_request": {
       "instance": { "id": 1 },
       "port": { "address": "COM42", "protocol": "serial" }
     }
   }
   ```
1. Open another terminal.
1. Run the following command in the new terminal to reinitialize the gRPC client instance:
   ```text
   $ grpcurl \
     -plaintext \
     -import-path ./rpc \
     -proto cc/arduino/cli/commands/v1/commands.proto \
     -d '{"instance": {"id": 1}}' \
     127.0.0.1:50051 \
     cc.arduino.cli.commands.v1.ArduinoCoreService.Init
   ```

🐛 The reinitialization hangs.

### Expected behavior

`cc.arduino.cli.commands.v1.ArduinoCoreService.Init` does not hang under any conditions.

### Arduino CLI version

863c1ec365af2cf90ee41f6fcadfa675e5d61704

### Operating system

- Linux
- macOS
- Windows

### Operating system version

- Ubuntu 24.04
- macOS Sonoma
- Windows 11

### Additional context

I bisected the regression to bb815cfe050d31ea827b7db3814bd24cf88049ee / https://github.com/arduino/arduino-cli/pull/2565 (the fault does not occur when using the build from the previous commit 917dcc5df6c206f6d1c6e93ec745fd931937cf3d).

---

Originally reported at https://github.com/arduino/arduino-cli/issues/2347#issuecomment-2378683212 (note the original issue report https://github.com/arduino/arduino-cli/issues/2347 was from long before the time of the regression, so is different from the Monitor-specific issue tracked here even if the symptoms are somewhat similar).

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

frankcohen, 2024-10-09:
Confirmed the issue on nightly build 2.3.2 20241001. Workaround is to close Serial Monitor. -Frank

closed 2024-10-11 by cmaglie

daxliniere, 2025-02-22:
This is not resolved. I am using v2.3.4 on Windows 10 and this is still a problem.

PR #2728 (2024-10-11): Fix monitor init handling
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [X] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [X] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` ha [... cut]