arduino/arduino-cli #2727: Excessive FQBN validation

grav, 2024-10-09:
### Describe the problem

The FQBN validation seems excessive.

I tried [disabling the check](https://github.com/arduino/arduino-cli/blob/master/internal/arduino/cores/fqbn.go#L65), and then I can succesfully list the board, and compile and upload to the board with my fork.

### To reproduce

```
$ arduino-cli board list
Error starting discovery: Invalid FQBN: fqbn's field dadamachines - M4 contains an invalid character
Port         Protocol Type              Board Name FQBN Core
/dev/ttyACM0 serial   Serial Port (USB) Unknown
```

### Expected behavior

```
$ ./arduino-cli board list
Port         Protocol Type              Board Name           FQBN                           Core
/dev/ttyACM0 serial   Serial Port (USB) dadamachines doppler dadamachines - M4:samd:doppler dadamachines - M4:samd
```

### Arduino CLI version

arduino-cli  Version: 1.0.3-arch Commit: 8b6ad258e88581a9e77b2eede65aeeeeb265008c Date:

### Operating system

Linux

### Operating system version

Linux 6.10.6-2-aarch64-ARCH aarch64

### Additional context

I'm not sure if it's a bug, or if the fqbn check is actually valuable. If so, maybe there should be a switch to ignore invalid characters? I'd be happy to try and implement it.

The specific board is not manufactured anymore, and I'm not sure if I can change the fqbn somehow.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

martin-eden, 2024-10-21:
I believe spaces are not allowed in [FQBN specification](https://arduino.github.io/arduino-cli/dev/FAQ/#whats-the-fqbn-string). Dada can use `_` instead.

grav, 2024-10-21:
> I believe spaces are not allowed in [FQBN specification](https://arduino.github.io/arduino-cli/dev/FAQ/#whats-the-fqbn-string). Dada can use `_` instead.

Yeah, looking again at the regex, it seems it's not the dash but the space that may be the problem, and it matches the specification, as you say.

The issue here is that the Doppler isn't really developed anymore. So I was thinking that a switch for backwards compatibility might be a good solution?

per1234, 2024-10-21:
Hi @grav. As @martin-eden explained, the problem is that @nevvkid used a vendor name that is not compliant with the specification:

https://github.com/dadamachines/arduino-board-index/blob/03e13338d4787c0a8eed89357c92cd7cf355e456/package_dadamachines-m4_index.json#L3

```json
"name": "dadamachines - M4",
```

We will not invest any resources into supporting non-compliant platforms. Please either submit a bug report or pull request to @dadamachines:

https://github.com/dadamachines/arduino-board-index/issues

---

> The issue here is that the Doppler isn't really developed anymore.

If @dadamachines has abandoned the project then you can make a fork of it.

Alternatively you can simply rename the platform's vendor folder to a compliant name (the vendor component of the FQBN is based on the folder name). The folder is at this path on your hard drive:

```text
/home/<username>/.arduino15/packages/dadamachines - M4
```

(Where `<username>` is your Linux username)

❗ The `.arduino15` folder may be hidden by default in your file manager and terminal.

closed 2024-10-21 by per1234

grav, 2024-10-22:
@per1234 Thanks for the reply.

> We will not invest any resources into supporting non-compliant platforms 

Understandable 👍 

I previouly tried to figure out if I could fix this on the Doppler side, but to no avail. I'll try your suggestion 👍