arduino/arduino-cli #2742: Boards with a board ID with a '/' can not be selected.

sterretjeToo, 2024-11-02:
### Describe the problem

Preamble:
Posted here as I suspect that this is a problem with the CLI and not with the IDE.

**Short:**
Unable to select the HTCC-AB02A in the Cubecell Development Framework 1.5.0; see  https://forum.arduino.cc/t/board-for-htcc-ab02a/1317566/
Root cause is a `/` in the board ID in boards.txt.

**Long:**
This problem was detected when a user installed CubeCell Development Framework using the boards manager in IDE 2.x and wanted to select the HTCC-AB02A board. I've confirmed the behaviour in both IDE 2.3.3 on Windows and IDE 1.8.19 on Windows.

If a different board was currently selected (e.g. Uno), the drop down (next to the debug button) in IDE 2.x does not reflect the change; neither does the status bar at the bottom. If no board was selected, the drop down shows "Select Board".

The issue was fixed by replacing all `CubeCell-1/2AA` board IDs by `CubeCell-1-2AA`. I only verified this for the IDE 2.3.3 installation on my system.

I did consult https://arduino.github.io/arduino-cli/0.34/platform-specification/#boardstxt to verifty whether there are any reserved characters and could not find it.

### To reproduce

1. Install the CubeCell Development Framework using the boards manager in IDE 2.x
2. Select the board CubeCell-1/2AA Node（HTCC-AB02A.

### Expected behavior

Be able to select the board. If the board can't be selected due to an error in boards.txt, an error message.

### Arduino CLI version

Sorry, nightly build arduino-ide_nightly-20241101_Windows_64bit

### Operating system

Windows

### Operating system version

Win11 Home

### Additional context

I've checked both https://github.com/arduino/arduino-cli and https://github.com/arduino/arduino-ide/issues and could not find a relevant issue.

The issue might be a bug in the CLI/IDE or a platform specification that is not 100% conclusive.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

sterretjeToo, 2024-11-02:
Note:
Original user is a Mac user so it also affects Mac users (and more than likely Linux Linux users).

per1234, 2024-11-02:
Hi @sterretjeToo. Unfortunately an inappropriate location was chosen for the documentation of the allowed characters for the board ID so I'm not surprised you couldn't find it:

https://arduino.github.io/arduino-cli/dev/FAQ/#whats-the-fqbn-string

>  Each field accepts letters (A-Z or a-z), numbers (0-9), underscores (_), dashes(-) and dots(.)

So this is a bug in the "CubeCell Development Framework" boards platform and should be reported to their issue tracker:

https://github.com/HelTecAutomation/CubeCell-Arduino/issues

closed 2024-11-02 by per1234

sterretjeToo, 2024-11-03:
Issue reported: https://github.com/HelTecAutomation/CubeCell-Arduino/issues/291