arduino/arduino-cli #2751: Preprocessor define to identify compiling with Arduino-IDE/CLI

pdppdp, 2024-11-10:
### Describe the request

I am unable to find a pre-processor defined symbol to unambiguously detect that the compilation is Arduino.

Every other compiler infrastructure I have used has one of these.  They are important for producing portable code.

It is trivial to add, can one please be added?

(Why doesn't Arduino have one already?)

### Describe the current behavior

There is no symbol, such as __ARDUINO__, __ARDUINO_IDE__ or __ARDUINO_CLI__ defined during compilation.

### Arduino IDE version

2.33

### Operating system

Windows

### Operating system version

windows 11 pro

### Additional context

This is not a new problem.  This question in a number of forms can be found in the arduino forum.  

A "recent" thread started by user:"stecoop" is a delightful example of a professional programmer with a simple need, asking a reasonable question, and getting a lot of useless non-answers from folks who don't understand the plainly stated question.  A whole lot of useless pain.
https://forum.arduino.cc/t/predefined-preprocessor-variables/1002170/11

One supposes this developer community has experienced compiler detection before.  Here are some examples gathered from my portable projects, just to demonstrate how widespread (and old) this practice is:
#ifdef  __BORLANDC__
#ifdef  _MSC_VER
#ifdef  __IBMC__
#ifdef  __GNUC__
#ifdef ESP_PLATFORM

One can see a somewhat "standard" industry pattern exists:
 <two underscores><your "name"><two underscores>

Following that, the new symbol could be __ARDUINO__.

I'm happy to join the fray and figure out where the "-D __ARDUINO__" goes, and make the change...

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

per1234, 2024-11-10:
Hi @pdppdp.

>  the new symbol could be ARDUINO.

This is already a common convention, defined by all official Arduino boards platforms:

- https://github.com/arduino/ArduinoCore-avr/blob/1.8.6/platform.txt#L54-L61
- https://github.com/arduino/ArduinoCore-mbed/blob/4.2.1/platform.txt#L72-L79
- https://github.com/arduino/ArduinoCore-megaavr/blob/1.8.8/platform.txt#L55-L62
- https://github.com/arduino/ArduinoCore-renesas/blob/1.2.2/platform.txt#L70-L80
- https://github.com/arduino/ArduinoCore-sam/blob/790ff2c852bf159787a9966bddee4d9f55352d15/platform.txt#L66-L73
- https://github.com/arduino/ArduinoCore-samd/blob/v1.6.1/platform.txt#L79-L86
- https://github.com/arduino/ArduinoCore-zephyr/blob/0bf61186fb17cc9a2f71e1d30b2a0f857708f68f/platform.txt#L72-L79

And also by all 3rd party platforms I am aware of, for example;

- https://github.com/esp8266/Arduino/blob/3.1.2/platform.txt#L130-L137
- https://github.com/espressif/arduino-esp32/blob/3.0.7/platform.txt#L150-L157
- https://github.com/MCUdude/MiniCore/blob/v3.0.2/avr/platform.txt#L55-L62
- https://github.com/MCUdude/MegaCore/blob/v3.0.2/avr/platform.txt#L55-L62
- https://github.com/SpenceKonde/ATTinyCore/blob/1.5.2/avr/platform.txt#L54-L61
- https://github.com/SpenceKonde/DxCore/blob/1.5.11/megaavr/platform.txt#L21
- https://github.com/adafruit/ArduinoCore-samd/blob/v1.6.1/platform.txt#L79-L86
- https://github.com/sparkfun/Arduino_Boards/blob/96a65e9e0cd4600792a1f5f4cff196e655de5ae8/sparkfun/avr/platform.txt#L53-L60
- https://github.com/sparkfun/Arduino_Boards/blob/96a65e9e0cd4600792a1f5f4cff196e655de5ae8/sparkfun/samd/platform.txt#L82-L89

So if you find a 3rd party platform that does not define this macro, please submit a bug report to the platform developers.

As for your proposal that the Arduino development tools should define the macro, we avoid any such thing because we want to give the platform developers maximum control (for example, some compilers use a different mechanism than a `-D` flag to define global macros).

It would be very beneficial for Arduino to provide formal documentation for the interfaces such as the `ARDUINO` macro that a platform is expected to provide, and we are tracking that task at https://github.com/arduino/arduino-cli/issues/985.

closed 2024-11-10 by per1234

pdppdp, 2024-11-14:
In a similar vein..

How does one detect compilation of a library?

I have a CPP library consisting of two files, one for the class def, one for the code itself.  Much like the Serial object, I wish a sole static instance.

So at the end of the header I have this:

```
//  try to do this once and only once, and not when compiling the library
#ifndef Z
    // declare it always!
    extern SlimJim Z;
    
    #ifndef     DONT_DEDEFINE_A_SLIMJIM
        #define     DONT_DEFINE_A_SLIMJIM
        // define and instantiate the sole instance
        SlimJim Z = SlimJim();
    #endif
#endif
```

When compiling the library I define DONT_DEDEFINE_A_SLIMJIM via compiler options.

Arduino provides no access to setting pre-processor symbols, so what defined symbol tells me I am in the compilation of the library .vs. the compilation of the .ino file?