arduino/arduino-cli #2755: The proto zip release asset does not contain the `google/rpc/status.proto` file for v1.1.0

dankeboy36, 2024-11-17:
### Describe the problem

The `google/rpc/status.proto` has been removed from the 1.1.0 release, although it is used (for example, `import "google/rpc/status.proto"`; in `cc/arduino/cli/commands/v1/commands.proto`), and downstream it breaks the proto to TS/JS generation. I believe the Arduino IDE will most likely also encounter this issue. I am about to release a new TS API for the CLI, and I am getting an error. It worked fine with the `1.0.4` release: https://www.npmjs.com/package/ardunno-cli/v/0.1.9?activeTab=code.

### To reproduce

Use curl/wget or another way to download the 1.0.4 proto zip and the 1.1.0 version. Unpack them and see the `status.proto` is not packed with the zip.

```
% unzip arduino-cli_1.1.0_proto.zip -d 1.1.0-out
Archive:  arduino-cli_1.1.0_proto.zip
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/commands.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/compile.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/upload.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/common.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/lib.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/port.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/monitor.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/core.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/board.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/settings.proto  
  inflating: 1.1.0-out/cc/arduino/cli/commands/v1/debug.proto  
% unzip arduino-cli_1.0.4_proto.zip -d 1.0.4-out
Archive:  arduino-cli_1.0.4_proto.zip
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/settings.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/monitor.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/core.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/compile.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/port.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/board.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/commands.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/common.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/upload.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/debug.proto  
  inflating: 1.0.4-out/cc/arduino/cli/commands/v1/lib.proto  
  inflating: 1.0.4-out/google/rpc/status.proto
```

Escaped command to reproduce:
```
"/Users/akos.kitta/Desktop/dev/ardunno-cli/node_modules/protoc/protoc/bin/protoc" "--plugin=/Users/akos.kitta/Desktop/dev/ardunno-cli/node_modules/ts-proto/protoc-gen-ts_proto" "--proto_path=/var/folders/g0/n_2cw4ds44l2byzv39xrc4340000gp/T/arduino-cli-17215-XcWyqFmh6gjw/rpc" "--ts_proto_opt=outputServices=nice-grpc,outputServices=generic-definitions,oneof=unions,useExactTypes=false,paths=source_relative,esModuleInterop=true,exportCommonSymbols=false,useOptionals=none" "--ts_proto_out=src/api" "cc/arduino/cli/commands/v1/board.proto" "cc/arduino/cli/commands/v1/commands.proto" "cc/arduino/cli/commands/v1/common.proto" "cc/arduino/cli/commands/v1/compile.proto" "cc/arduino/cli/commands/v1/core.proto" "cc/arduino/cli/commands/v1/debug.proto" "cc/arduino/cli/commands/v1/lib.proto" "cc/arduino/cli/commands/v1/monitor.proto" "cc/arduino/cli/commands/v1/port.proto" "cc/arduino/cli/commands/v1/settings.proto" "cc/arduino/cli/commands/v1/upload.proto"
```

### Expected behavior

`google/rpc/status.proto` is part of the release proto artifact

### Arduino CLI version

1.1.0

### Operating system

N/A

### Operating system version

n/a

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

cmaglie, 2024-11-18:
We recently migrated all protobuf tooling to `buf`: https://github.com/arduino/arduino-cli/pull/2736

One of the nice features of `buf` is that it can handle the "dependencies" automatically https://github.com/arduino/arduino-cli/blob/master/buf.yaml#L3, so we don't have to mirror the `status.proto` anymore. The downside is that we do not package it anymore.

We may download and package it in some way @alessio-perugini

closed 2024-11-21 by cmaglie

per1234, 2024-12-19:
Hi @dankeboy36. I don't have an answer to your specific question, but I can offer an alternative solution, which is to switch from the excellent [**grpcurl**](https://github.com/fullstorydev/grpcurl) to using [**Buf CLI**](https://buf.build/product/cli), which is now the preferred tool for the Arduino CLI project. **Buf CLI** will take care of providing this missing dependency for you.

Here is the equivalent command using **Buf CLI**:

```text
npx @bufbuild/buf \
  curl \
    --http2-prior-knowledge \
    --protocol grpc \
    --schema ./rpc \
    http://localhost:50051/cc.arduino.cli.commands.v1.ArduinoCoreService/Create
```

alessio-perugini, 2024-12-19:
@dankeboy36 
```bash
# Assume you're in the `arduino-cli` folder

$ buf export . -o /tmp/test
$ tree /tmp/test

/tmp/test
├── cc
│   └── arduino
│       └── cli
│           └── commands
│               └── v1
│                   ├── board.proto
│                   ├── commands.proto
│                   ├── common.proto
│                   ├── compile.proto
│                   ├── core.proto
│                   ├── debug.proto
│                   ├── lib.proto
│                   ├── monitor.proto
│                   ├── port.proto
│                   ├── settings.proto
│                   └── upload.proto
└── google
    └── rpc
        └── status.proto

8 directories, 12 files
```

This will generate all the dependencies, and you can continue to use grpcurl if you wish 😄