arduino/arduino-cli #2763: Detect libraries in Sketch Folder

fedy0, 2024-11-22:
### Describe the request

The sketch folder structure described in the [arduino-cli documentation](https://arduino.github.io/arduino-cli/1.0/sketch-specification/#sketch-file-structure-example) does not work (i.e. the libraries added in the sketch folder can not be found by both the arduino-cli and the arduino-builder
[image]

### Describe the current behavior

It returns the error below when one tries to compile a sketch with a library whose location is within the sketch folder itself
```bash
> fatal error: SomeLib.h: No such file or directory
```

### Arduino CLI version

1.1.1

### Operating system

Linux

### Operating system version

24.04

### Additional context

If the build procedure can be made in such a way that the build process (both the cli & builder) checks the sketch folder for libraries first before the `Arduino/libraries` and `/home/<user>/.arduino15/packages/<platform>/hardware/<core>/<version>/libraries` locations that would be great and in adherence to the [sketch folder structure specification](https://arduino.github.io/arduino-cli/1.0/sketch-specification/#sketch-file-structure-example)

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My request contains all necessary details

fedy0, 2024-11-22:
Additionally, this feature can also be added in `sketch.yaml` file as show below. This will enable the addition of third party libraries or self developed libraries.
```
profiles:
  <PROFILE_NAME>:
    notes: <USER_NOTES>
    fqbn: <FQBN>
    programmer: <PROGRAMMER>
    platforms:
      - platform: <PLATFORM> (<PLATFORM_VERSION>)
        platform_index_url: <3RD_PARTY_PLATFORM_URL>
      - platform: <PLATFORM_DEPENDENCY> (<PLATFORM_DEPENDENCY_VERSION>)
        platform_index_url: <3RD_PARTY_PLATFORM_DEPENDENCY_URL>
    libraries:
      - <LIB_NAME> (<LIB_VERSION>)
      - <LIB_NAME> (<LIB_VERSION>)
        library_url: <3RD_PARTY_LIBRARY_DEPENDENCY_URL>
      - <LIB_NAME> (<LIB_VERSION>)
        library_url: <3RD_PARTY_LIBRARY_DEPENDENCY_URL>
    port: <PORT_NAME>
    port_config:
      <PORT_SETTING_NAME>: <PORT_SETTING_VALUE>
      ...
    protocol: <PORT_PROTOCOL>
  ...more profiles here...
```

per1234, 2024-11-22:
Hi @fedy0.

> ```text
> fatal error: SomeLib.h: No such file or directory
> ```

Even though the example in the specification shows the common use case of bundling a library's files with a sketch, the files in the `src` subfolder of the sketch are not treated as libraries. They are only additional files of the sketch. For this reason, you must use the relative path to the file in the `#include` directive. In the example shown in the specification, if you wanted to add an `#include` directive for `SomeLib.h` to `Foo.ino`, it would look like this:

```cpp
#include "src/SomeLib/src/SomeLib.h
```

> the build procedure can be made in such a way that the build process (both the cli & builder) checks the sketch folder for libraries

I see we have another request for this feature at https://github.com/arduino/arduino-cli/issues/1255.

> this feature can also be added in `sketch.yaml` file as show below

I see we have another request for this feature at https://github.com/arduino/arduino-cli/issues/2133

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the previous feature requests.

closed 2024-11-22 by per1234

fedy0, 2024-11-26:
Hi

The relative path doesn't work that's why I opened this request.

Nevertheless, thank you for your prompt response.