arduino/arduino-cli #2778: Package Specification Yielding "platform is not available for your OS" Despite Package File Specifications OK for OS (Linux)

haydenth, 2024-12-07:
### Describe the problem

I've created a packages file visible below. Or you can find here:
```
https://raw.githubusercontent.com/atmosphericiq/aiq-board-defs/refs/heads/main/package_atmosphericiq_index.json
```

```
{
  "packages": [
    {
      "name": "atmosphericiq",
      "maintainer": "Atmospheric IQ",
      "websiteURL": "https://github.com/atmosphericiq/aiq-board-defs",
      "email": "thayden@engineeringdirector.com",
      "platforms": [
        {
          "name": "AtmosphericIQ CBOT Boards",  
          "architecture": "stm32", 
          "version": "1.0.0", 
          "category": "Contributed",
          "url": "https://github.com/atmosphericiq/aiq-board-defs/archive/refs/tags/v1.0.0.zip",
          "archiveFileName": "v1.0.0.zip", 
          "checksum": "SHA-256:328fcd7a9b3eca5893c3a385cac629ae942eb5afeef353dcd1acc7102e09a155",
          "size": "24638", 
          "boards": [
            {
              "name": "CBOT_V9"
            },
            {
              "name": "CBOT_V11"
            }
          ],
          "toolsDependencies": [
             {
              "packager": "STMicroelectronics",
              "name": "xpack-arm-none-eabi-gcc",
              "version": "13.2.1-1.1"
            }
          ]
        }
      ],
      "tools": [
                {
          "name": "xpack-arm-none-eabi-gcc",
          "version": "13.2.1-1.1",
          "systems":
          [
            {
              "host": "arm-linux-gnueabihf",
              "url": "https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v13.2.1-1.1/xpack-arm-none-eabi-gcc-13.2.1-1.1-linux-arm.tar.gz",
              "archiveFileName": "xpack-arm-none-eabi-gcc-13.2.1-1.1-linux-arm.tar.gz",
              "checksum": "SHA-256:9a6db147c34f7ea668cc37a139d2667a58f8b2bbee2359f23e48ffd300f8fc2f",
              "size": "257683778"
            },
            {
              "host": "aarch64-linux-gnu",
              "url": "https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v13.2.1-1.1/xpack-arm-none-eabi-gcc-13.2.1-1.1-linux-arm64.tar.gz",
              "archiveFileName": "xpack-arm-none-eabi-gcc-13.2.1-1.1-linux-arm64.tar.gz",
              "checksum": "SHA-256:ab7f75d95ead0b1efb7432e7f034f9575cc3d23dc1b03d41af1ec253486d19de",
              "size": "265190134"
            },
            {
              "host": "i686-mingw32",
              "url": "https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v13.2.1-1.1/xpack-arm-none-eabi-gcc-13.2.1-1.1-win32-x64.zip",
              "archiveFileName": "xpack-arm-none-eabi-gcc-13.2.1-1.1-win32-x64.zip",
              "checksum": "SHA-256:56b18ccb0a50f536332ec5de57799342ff0cd005ca2c54288c74759b51929e4f",
              "size": "306340878"
            },
            {
[... 100 lines cut ...]
INFO[0000] Running from console, will run pre-uninstall by default
Error during install: Platform 'atmosphericiq:stm32' not found: platform is not available for your OS
```

### Expected behavior

I would expect it to install both the `xpack-arm-none-eabi-gcc-13.2.1-1.1-linux-x64.tar.gz` and the package itself.

### Arduino CLI version

arduino-cli  Version: nightly-20241207 Commit: 6dbff9f Date: 2024-12-07T01:34:43Z

### Operating system

Linux

### Operating system version

Linux stardust 6.8.0-49-generic #49~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov  6 17:42:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

haydenth, 2024-12-07:
cc @fronders

per1234, 2024-12-08:
Hi @haydenth. The `packages[*].platforms[*].toolsDependencies` key in your package index specifies a dependency on version `13.2.1-1.1` of a tool with ID `xpack-arm-none-eabi-gcc`, which is provided by the package with ID `STMicroelectronics` (in shorthand, `STMicroelectronics:xpack-arm-none-eabi-gcc@13.2.1-1.1`).

You define a tool with that ID and version in the package with ID `atmosphericiq` (in shorthand, `atmosphericiq:xpack-arm-none-eabi-gcc@13.2.1-1.1`), but that is not the same as the specified `STMicroelectronics:xpack-arm-none-eabi-gcc@13.2.1-1.1` dependency. So when Arduino CLI goes looking for the `STMicroelectronics:xpack-arm-none-eabi-gcc@13.2.1-1.1` dependency, it can't find it, and thus the boards platform installation fails.

So you should be able to fix the bug with this patch:

```diff
--- a/package_atmosphericiq_index.json
+++ b/package_atmosphericiq_index.json
@@ -25,7 +25,7 @@
           ],
           "toolsDependencies": [
              {
-              "packager": "STMicroelectronics",
+              "packager": "atmosphericiq",
               "name": "xpack-arm-none-eabi-gcc",
               "version": "13.2.1-1.1"
             }
```

You can learn more about this subject from the Arduino Package Index Specification:

https://arduino.github.io/arduino-cli/latest/package_index_json-specification/

If you have any questions or problems, feel free to post over on Arduino Forum:

https://forum.arduino.cc/

closed 2024-12-08 by per1234

haydenth, 2024-12-08:
Thank you SO MUCH for your kind reply here. This absolutely solved my problem.