arduino/arduino-cli #2784: `board list --watch` fails to detect the built-in Bluetooth port on macOS

dankeboy36, 2024-12-16:
### Describe the problem

I have been experiencing an intermittent issue with the `board list --watch` command while using version 1.1.1 of the CLI on macOS. Occasionally, the command does not detect the built-in Bluetooth port, resulting in missing watch events via gRPC.

### To reproduce

1. Open a Terminal on macOS.
2. Run the command: `board list --watch`
3. Monitor for the detection of the built-in Bluetooth port.
4. Press Ctrl+C to terminate the watch process.
5. Repeat from 1.) until you hit the bug.

```sh
akos.kitta@prprpr ~ % "/Users/akos.kitta/Library/Application Support/Code/User/globalStorage/undefined_publisher.vscode-ardunno/install/v1.1.1/arduino-cli" board list --watch
Port Type Event Board Name FQBN Core

/dev/cu.Bluetooth-Incoming-Port Serial Port Connected

^C
akos.kitta@prprpr ~ % "/Users/akos.kitta/Library/Application Support/Code/User/globalStorage/undefined_publisher.vscode-ardunno/install/v1.1.1/arduino-cli" board list --watch
Port Type Event Board Name FQBN Core

/dev/cu.Bluetooth-Incoming-Port Serial Port Connected

^C
akos.kitta@prprpr ~ % "/Users/akos.kitta/Library/Application Support/Code/User/globalStorage/undefined_publisher.vscode-ardunno/install/v1.1.1/arduino-cli" board list --watch
Port Type Event Board Name FQBN Core

/dev/cu.Bluetooth-Incoming-Port Serial Port Connected

^C
akos.kitta@prprpr ~ % "/Users/akos.kitta/Library/Application Support/Code/User/globalStorage/undefined_publisher.vscode-ardunno/install/v1.1.1/arduino-cli" board list --watch
Port Type Event Board Name FQBN Core

/dev/cu.Bluetooth-Incoming-Port Serial Port Connected

^C
akos.kitta@prprpr ~ % "/Users/akos.kitta/Library/Application Support/Code/User/globalStorage/undefined_publisher.vscode-ardunno/install/v1.1.1/arduino-cli" board list --watch
Port Type Event Board Name FQBN Core

```

The OS can detect the serial port with `ls /dev/tty.*` at the same time while the watch process runs

Proofs 👇 

[image]

[image]

### Expected behavior

The `board list --watch` command should consistently detect the built-in Bluetooth port and trigger appropriate board list watch events.

### Arduino CLI version

1.1.1

### Operating system

macOS

### Operating system version

11.7.10

### Additional context

- I initially suspected a potential bug in my code via grpc, but I could reproduce it from a shell.
- I have never noticed this defect before 1.1.1.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

cmaglie, 2025-01-21:
I tried to reproduce it on my Mac, I've launched the CLI a hundred times, and never failed :-/.

Can I ask you to bisect and find the wrong commit?
Is your Mac x86 or arm?

dankeboy36, 2025-01-21:
Thank you for investigating the issue.

> Is your Mac x86 or arm?

I encountered the bug on an Intel Mac, but I will check if I can reproduce it on a Mac with an ARM chip.

> Can I ask you to bisect and find the wrong commit?

Sure. It's not my highest priority at the moment, but when I can reproduce it, I will update the thread.

Also, IDE users have experienced the same problem on Linux: https://github.com/arduino/arduino-ide/issues/2600.