arduino/arduino-cli #2796: Option to disable "Used Library" and "Used Platform" output  when compiling

obra, 2024-12-24:
### Describe the request

When building a single sketch, the "Used Library" and "Used Platform" output from arduino-cli are great, but when building a test suite, it can somewhat clutter STDOUT.

`--quiet` does not suppress that output
`--show-properties=disabled` does not suppress that output.

It would be delightful to have a command-line option to suppress that output.

I could probably cobble together some code as a proposed PR if a maintainer provided some guidance about whether a good implementation was likely to be accepted and what the name of the option should be / what the right implementation strategy was.

### Describe the current behavior

(To the best of my knowledge, and based on my not-so-great go skills) It is not currently possible to disable the printing of the after-`arduino-cli compile` build summary that looks something like this:

```

Used library                             Version Path
Kaleidoscope                             0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv
Kaleidoscope-Hardware-Keyboardio-Model01 0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-Hardware-Keyboardio-Model01
KeyboardioHID                            0.0.1   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/KeyboardioHID
EEPROM                                   2.0     /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/.arduino/user/hardware/keyboardio/avr/libraries/EEPROM
Kaleidoscope-EEPROM-Settings             0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-EEPROM-Settings
Kaleidoscope-EEPROM-Keymap               0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-EEPROM-Keymap
Kaleidoscope-FocusSerial                 0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-FocusSerial
Kaleidoscope-FirmwareVersion             0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-FirmwareVersion
Kaleidoscope-MouseKeys                   0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-MouseKeys
Kaleidoscope-Macros                      0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-Macros
Kaleidoscope-MacroSupport                0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-MacroSupport
Kaleidoscope-Ranges                      0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-Ranges
Kaleidoscope-DynamicMacros               0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-DynamicMacros
Kaleidoscope-NumPad                      0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-NumPad
Kaleidoscope-LEDEffect-BootGreeting      0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-LEDEffect-BootGreeting
Kaleidoscope-LEDEffect-SolidColor        0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-LEDEffect-SolidColor
Kaleidoscope-LEDEffect-Breathe           0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-LEDEffect-Breathe
Kaleidoscope-LEDEffect-Chase             0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-LEDEffect-Chase
Kaleidoscope-LEDEffect-Rainbow           0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-LEDEffect-Rainbow
Kaleidoscope-LED-Palette-Theme           0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-LED-Palette-Theme
Kaleidoscope-Colormap                    0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-Colormap
Kaleidoscope-HostPowerManagement         0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-HostPowerManagement
Kaleidoscope-MagicCombo                  0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-MagicCombo
Kaleidoscope-Qukeys                      0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-Qukeys
Kaleidoscope-USB-Quirks                  0.0.0   /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/plugins/Kaleidoscope-USB-Quirks

Used platform  Version Path
keyboardio:avr 1.6.11  /Users/jesse/Documents/GitHub/keyboardio/firmware/priv/.arduino/user/hardware/keyboardio/avr
```

### Arduino CLI version

1.1.1

### Operating system

macOS

### Operating system version

15.2

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My request contains all necessary details

Tyler-Barnes, 2025-01-03:
I would also really like this option. The information I want to see at the end of the compilation is the memory usage. This extra wall of output just pushes that off screen and tells me information I already know.

cmaglie, 2025-01-21:
A possible fix in #2820.

> I would also really like this option. The information I want to see at the end of the compilation is the memory usage. This extra wall of output just pushes that off screen and tells me information I already know.

@Tyler-Barnes 
Actually, in the PR I made it completely silent, also the information about code size is silenced. Do you think I should re-enable it?

@obra WDYT?

obra, 2025-01-21:
Awesome. Thank you! That looks great for my purposes. (I actually like the memory usage not printing in --quiet mode)

cmaglie, 2025-01-21:
@obra 
I see that many users want to keep the two lines about memory usage. Would re-enabling them be a good compromise for you too?

Also, did you know about this GH action for CI builds: https://github.com/arduino/compile-sketches

obra, 2025-01-21:
I feel like `--quiet` should be...quiet and not print a report about what just happened unless something exceptional happened. That's the *usual* standard for command-line tools.

Memory usage lines and no libraries report is better than memory usage lines and a library report. I'm not going to kick and scream too loudly if I can't get *exactly* what I want.  Although, I'd love to find a way to *not* have the memory usage lines. 

I presume Arduino's preference is that the 'default' output remains the same as it is today and that you not complicate the output options too much. 

I thought about whether the log level could get used to tease these bits out output apart and I don't *think* there's a good way to do that.

I do wonder if the middle-ground here might be "Show the build size/memory usage report If we're building with --quiet and the build is too big or uses too much memory" - At that point, it's rightly a warning?

cmaglie, 2025-01-22:
> I do wonder if the middle-ground here might be "Show the build size/memory usage report If we're building with --quiet and the build is too big or uses too much memory" - At that point, it's rightly a warning?

but this won't cover the common case where the dev is keeping an eye on the sketch size. Maybe the best compromise is to remove the libraries recap in the default case, something like this:

Command line | A successful build shows | A build with errors shows
-------------------|-------------|-------------
`arduino-cli compile` | Sketch size | Sketch size, used libraries
`arduino-cli compile -v` | Build commands, sketch size, used libraries | Build commands, sketch size, used libraries
`arduino-cli compile -q` | nothing | Errors only

Tyler-Barnes, 2025-01-22:
My grip is only that the libraries are at the end. Just put them above the size information and I'm happy. I like verbose output, but I don't like the last thing to be libraries pushing the size information up. If I can only remove them with -quiet then I'm still in the same spot.

cmaglie, 2025-01-24:
> My grip is only that the libraries are at the end. Just put them above the size information and I'm happy. I like verbose output, but I don't like the last thing to be libraries pushing the size information up. If I can only remove them with -quiet then I'm still in the same spot.

Unfortunately moving the libraries recap before the sketch size requires a lot more work because it needs a general improvement on the compiler output logging system. I've in it my task list but it's low priority for now.

By the way, the change proposed in #2820 is already a good improvement, so I'm going to merge that for the next release.

closed 2025-01-24 by cmaglie