arduino/arduino-cli #2832: Using `--input-file` with `arduino-cli upload` fails because it looks for a file name different from the one supplied

Kabouik, 2025-02-06:
### Describe the problem

I need to upload precompiled binaries but I am facing the following issue:

### To reproduce

```
$ arduino-cli upload --port /dev/ttyUSB0 --fqbn Heltec-esp32:esp32:heltec_wireless_stick_lite_V3 --input-file /path/to/my/precompiled/files//lora_lte_node_V2.15_v1_40MHz.bin`

usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/path/to/my/precompiled/files/lora_lte_node_V2.15_v1_40MHz.bootloader.bin'
Failed uploading: uploading error: exit status 2
```
Note how the filename differs in the error and in the command I used. If I copy the file to a new name to add the missing "bootloader" part that `arduino-cli` expects, then it tries to add this suffix again (i.e., "bootloader.bootloader").

What am I doing wrong?

### Expected behavior

The supplied binary filename should be used to upload to the board.

### Arduino CLI version

Version: 0.35.3 Commit: 95cfd654 Date: 2024-02-19T13:24:24Z

### Operating system

Linux

### Operating system version

Debian 6.11.11 distrobox

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

Kabouik, 2025-02-06:
Using `esptool` to flash just the .bin file without the `bootloader`, partition`and `boot_app0` counterparts seems to do what I want:

```
python3 "$HOME/.arduino15/packages/Heltec-esp32/tools/esptool_py/4.6/esptool.py" --chip esp32s3 --port "/dev/ttyUSB0" --baud 921600  --before default_reset --after hard_reset write
_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x10000 "./lora_lte_node_V2.15_v1_40MHz.bin"
```

I'm not positive that I'm using the correct options though.

per1234, 2025-02-07:
Thanks for your report @Kabouik.

>`arduino-cli upload`'s `--input-file` argument being singular, also in the help for this command, suggested that only one file was necessary, but maybe I am misusing it.

Yes, this is definitely misleading. The documentation needs to be corrected:

https://github.com/arduino/arduino-cli/blob/a2eebcd805aca6a9c247192854eaaf77628094c9/internal/cli/upload/upload.go#L75

Despite what the documentation claims, Arduino CLI does not necessarily upload the file at the path passed to the flag. What it does instead is extract two separate things from that path, which are then provided to the boards platform for use in the `upload` (or `program` if the `--programmer` flag is also passed) [pattern](https://arduino.github.io/arduino-cli/dev/platform-specification/#tools) (the template from which the upload command line is generated):

- [Parent folder](https://github.com/arduino/arduino-cli/blob/a2eebcd805aca6a9c247192854eaaf77628094c9/commands/service_upload.go#L794) -> [`build.path`](https://github.com/arduino/arduino-cli/blob/a2eebcd805aca6a9c247192854eaaf77628094c9/commands/service_upload.go#L457)
- [Filename without extension](https://github.com/arduino/arduino-cli/blob/a2eebcd805aca6a9c247192854eaaf77628094c9/commands/service_upload.go#L794) -> [`build.project_name`](https://github.com/arduino/arduino-cli/blob/a2eebcd805aca6a9c247192854eaaf77628094c9/commands/service_upload.go#L458)

There is no provision in the [Arduino boards platform framework](https://arduino.github.io/arduino-cli/dev/platform-specification/) for explicitly uploading a single binary file. The platform developers will configure their `upload` pattern with the expectation that it is used in combination with the files generated by `arduino-cli compile`, and thus the pattern might reference multiple binary files. That is the case with the `esp32:esp32` platform:

https://github.com/espressif/arduino-esp32/blob/3.1.1/platform.txt#L288-L289

```text
tools.esptool_py.upload.pattern_args=--chip {build.mcu} --port "{serial.port}" --baud {upload.speed} {upload.flags} --before default_reset --after hard_reset write_flash {upload.erase_cmd} -z --flash_mode keep --flash_freq keep --flash_size keep {build.bootloader_addr} "{build.path}/{build.project_name}.bootloader.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin" 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" 0x10000 "{build.path}/{build.project_name}.bin" {upload.extra_flags}
tools.esptool_py.upload.pattern="{path}/{cmd}" {upload.pattern_args}
```

When you compile a sketch for a board of the `esp32:esp32` platform, it will generate all the files the `upload` pattern references and place them in the `{build.path}`. So `arduino-cli upload --input-file` should work fine (even if not in the way the user expects) when passed the path of the primary binary file as generated by `arduino-cli compile`. But this may not be the case if you are trying to upload an arbitrary binary (which is the likely use case for this flag).

Unless the provided binary (or group of binaries) were specifically intended for use with `arduino-cli upload --input-file`, you should instead use the upload tool directly instead of using **Arduino CLI**, as you did with `esptool`.

cmaglie, 2025-02-07:
@Kabouik maybe the `--output-dir` flag in compile and the `--input-dir` in upload may be better for your use case?

```
$ arduino-cli compile -h | grep output-
      --output-dir string                     Save build artifacts in this directory.
$ arduino-cli upload -h | grep input
      --input-dir string              Directory containing binaries to upload.
  -i, --input-file string             Binary file to upload.
```

If you compile with the `--output-dir PATH` flag set, the artifacts needed for the upload will be saved in the given PATH, you can provide the same PATH with `--input-dir` to upload.