arduino/arduino-cli #2835: arduino-cli config dump returns nothing

mvandere, 2025-02-12:
### Describe the problem

arduino-cli config dump 
returns nothing
or at least just
{}
This is in version 1.1.1, It definitely worked in an earlier Arduino r2 release, but I can't find a way to download earlier versions to work out when it broke.

### To reproduce

As per above

### Expected behavior

Can't get a working version so I can't say exactly, but it use to definitely return
data: (path to arduino15 folder)
user: (path to default user save folder)
as well as other things

### Arduino CLI version

1.1.1

### Operating system

Windows

### Operating system version

10

### Additional context

BTW I couldn't check against the nightly build as my browser times out trying to access it.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-02-12:
Hi @mvandere. This was an intentional change which was introduced in Arduino CLI 1.0.0. It is documented in the migration guide for that release:

https://arduino.github.io/arduino-cli/dev/UPGRADING/#config-dump-no-longer-returns-default-configuration-values

As explained there, it is now necessary use `config get` to get the effective value of configuration keys which are being set implicitly via the defaults.

closed 2025-02-12 by per1234

Perlovka, 2026-04-10:
Yeah, let's drop usable function, so users should search for available values somewhere!
Arduino devs ability to shoot user feets has no limits 🤦