arduino/arduino-cli #2859: gRPC Documentation is missing

ChiefBacon, 2025-03-09:
### Describe the problem

The [integrations options page](https://github.com/arduino/arduino-cli/blob/master/docs/integration-options.md) links to [a nonexistent gRPC documentation page.](https://github.com/arduino/arduino-cli/blob/master/docs/rpc/commands.md) I don't know if I have just been looking in the wrong place, but I can not find any other detailed information about the gRPC connection. Thanks :)

### To reproduce

N/A

### Expected behavior

N/A

### Arduino CLI version

N/A

### Operating system

N/A

### Operating system version

N/A

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-03-09:
Hi @ChiefBacon. The gRPC documentation is generated from the source code, so we don't store it in the standard repository files. For this reason, the link doesn't work if you view the documentation files via the GitHub repository. You should instead view the documentation via the documentation website, here:

https://arduino.github.io/arduino-cli/latest/

It would be interesting to investigate whether the links to the gRPC documentation (and the links to the command line documentation) could be changed to work even when viewing the documentation files via the GitHub repository, without impacting the functionality of the website (and the website generation infrastructure).

ChiefBacon, 2025-03-09:
Ah, yes, I see! Thanks for pointing that out, and this doc helps a lot. I should be able to figure it out from here, thanks!

closed 2025-03-09 by ChiefBacon