arduino/arduino-cli #2861: gRPC server crashes on new sketch request

4ntoine, 2025-03-11:
### Describe the problem

I'm creating a new sketch with arduino-cli v1.2.0 and daemon crashes.
See "to reproduce" on the exact steps.

### To reproduce

1. Check out v1.2.0 tag
2. go build
3. ./arduino-cli daemon -v
4. See the crash:
```
~/dev/src/arduino-cli tags/v1.2.0 $./arduino-cli daemon -v
INFO[0000] arduino-cli version 0.0.0-git                
INFO[0000] Using config file: /Users/anton/Library/Arduino15/arduino-cli.yaml 
INFO[0000] Executing `arduino-cli daemon`               
Daemon is now listening on 127.0.0.1:50051
{"IP":"127.0.0.1","Port":"50051"}

INFO[0039] Checking signature                            index=/Users/anton/Library/Arduino15/package_index.json signatureFile=/Users/anton/Library/Arduino15/package_index.json.sig trusted=true
INFO[0039] Loading hardware from: /Users/anton/Library/Arduino15/packages 
INFO[0039] Loading package builtin from: /Users/anton/Library/Arduino15/packages/builtin 
INFO[0039] Checking existence of 'tools' path: /Users/anton/Library/Arduino15/packages/builtin/tools 
INFO[0039] Loading tools from dir: /Users/anton/Library/Arduino15/packages/builtin/tools 
INFO[0039] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0039] Loaded tool                                   tool="builtin:dfu-discovery@0.1.2"
INFO[0039] Loaded tool                                   tool="builtin:mdns-discovery@1.0.9"
INFO[0039] Loaded tool                                   tool="builtin:serial-discovery@1.4.1"
INFO[0039] Loaded tool                                   tool="builtin:serial-monitor@0.14.1"
INFO[0039] Loading libraries index file                  index=/Users/anton/Library/Arduino15/library_index.json
INFO[0039] Adding libraries dir                          dir=/Users/anton/Library/Arduino15/libraries isSingleLibrary=false location=ide
INFO[0039] Adding libraries dir                          dir=/Users/anton/Documents/Arduino/libraries isSingleLibrary=false location=user
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x2 addr=0x0 pc=0x1009dbd3c]

goroutine 8 [running]:
github.com/arduino/go-paths-helper.(*Path).Join(0x0, {0x140000df840?, 0x1?, 0x1005cc908?})
	/Users/anton/go/pkg/mod/github.com/arduino/go-paths-helper@v1.12.1/paths.go:92 +0x3c
github.com/arduino/arduino-cli/commands.(*arduinoCoreServerImpl).NewSketch(0x101b4d020?, {0x101453380?, 0x140000908b8?}, 0x140067d6640)
	/Users/anton/dev/src/arduino-cli/commands/service_sketch_new.go:58 +0x100
github.com/arduino/arduino-cli/rpc/cc/arduino/cli/commands/v1._ArduinoCoreService_NewSketch_Handler({0x101453380, 0x140001a8db0}, {0x101474168, 0x140067d2e40}, 0x14005bab080, 0x0)
	/Users/anton/dev/src/arduino-cli/rpc/cc/arduino/cli/commands/v1/commands_grpc.pb.go:1279 +0x1c0
google.golang.org/grpc.(*Server).processUnaryRPC(0x140001b0c00, {0x101474168, 0x140067d2db0}, 0x140001b93e0, 0x1400044c300, 0x101b64d48, 0x0)
	/Users/anton/go/pkg/mod/google.golang.org/grpc@v1.70.0/server.go:1400 +0xc9c
google.golang.org/grpc.(*Server).handleStream(0x140001b0c00, {0x101474ad0, 0x140000d4000}, 0x140001b93e0)
	/Users/anton/go/pkg/mod/google.golang.org/grpc@v1.70.0/server.go:1810 +0x900
google.golang.org/grpc.(*Server).serveStreams.func2.1()
	/Users/anton/go/pkg/mod/google.golang.org/grpc@v1.70.0/server.go:1030 +0x84
created by google.golang.org/grpc.(*Server).serveStreams.func2 in goroutine 6
	/Users/anton/go/pkg/mod/google.golang.org/grpc@v1.70.0/server.go:1041 +0x138
~/dev/src/arduino-cli tags/v1.2.0 $git log -1
commit 9c495211bba5543af0d49957d6c6982d18d1eabe (HEAD, tag: v1.2.0, origin/1.2.x, master)
Author: Cristian Maglie <c.maglie@arduino.cc>
Date:   Mon Feb 24 16:52:58 2025 +0100

    [skip-changelog] Investigate some integration-test failures (increase debugging prints) (#2844)
```

Looking into "service_sketch_new.go" line 58 i can see:
> sketchDirPath := paths.New(sketchesDir).Join(req.GetSketchName())

Where `sketchesDir = s.settings.GetString("directories.User")` as i did not pass sketches dir in the request.

Here is the trivial test (Dart syntax):
```
final channel = _getChannel();
    final client = ArduinoCoreServiceClient(channel);
    var createResponse = await client.create(CreateRequest.getDefault());
    await for (var res
        in client.init(InitRequest(instance: createResponse.instance))) {
      expect(res.hasError(), false, reason: res.toDebugString());
    }
    final newResponse =
        await client.newSketch(NewSketchRequest(sketchName: 'NewSketch2'));
    print('New sketch path is: ${newResponse.mainFile}');
    await channel.shutdown();
```

I suspect `s.settings.GetString("directories.User")` returns nothing and probably it's just my SDK not configured.
Happy to try something.

### Expected behavior

I get the response with new sketch path.
The arduino-cli daemon does not crash.

### Arduino CLI version

v1.2.0

### Operating system

macOS

### Operating system version

Sequioa 15.3

### Additional context

Locally built arduino-cli. Also tried installed one with `brew`.

Tested `b9edb782a265b99878b4bce489c1751c4dcbee61` locally instead of nightly.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

closed 2025-03-12 by alessio-perugini

alessio-perugini, 2025-03-12:
@4ntoine Thank you for submitting the issue, it has been fixed by #2862 . Ideally, we want to fix another issue before tagging the next release `v1.2.1`.

alessio-perugini, 2025-04-22:
@4ntoine we have finally released the fix in https://github.com/arduino/arduino-cli/releases/tag/v1.2.2