arduino/arduino-cli #2878: Boards from --additional-urls are missing on raspberry pi

omicronns, 2025-04-05:
### Describe the problem

STM32 boards does not appear in `board search` when running on raspberry pi.

### To reproduce

```
go build
./arduino-cli board search --additional-urls https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json
```

No STM32 boards are listed, when running this on raspberry pi, same exact steps work on my pc.

### Expected behavior

STM32 boards being available.

### Arduino CLI version

commit: 00a33b7466617ac6e4d9693a48cfe40d6a8efec8

### Operating system

Linux

### Operating system version

Manjaro ARM

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

cmaglie, 2025-04-07:
Could you copy the output of the command?

omicronns, 2025-04-07:
Sure, on RPi:
```
./arduino-cli board search --additional-urls https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

Downloading index: library_index.tar.bz2 downloaded                                                                                                                                                                                                                                      
Downloading index: package_index.tar.bz2 downloaded                                                                                                                                                                                                                                      
Downloading index: package_stmicroelectronics_index.json downloaded                                                                                                                                                                                                                      
Downloading missing tool builtin:mdns-discovery@1.0.9...
builtin:mdns-discovery@1.0.9 downloaded                                                                                                                                                                                                                                                  
Installing builtin:mdns-discovery@1.0.9...
Skipping tool configuration....
builtin:mdns-discovery@1.0.9 installed
Downloading missing tool builtin:serial-discovery@1.4.1...
builtin:serial-discovery@1.4.1 downloaded                                                                                                                                                                                                                                                
Installing builtin:serial-discovery@1.4.1...
Skipping tool configuration....
builtin:serial-discovery@1.4.1 installed
Downloading missing tool builtin:serial-monitor@0.15.0...
builtin:serial-monitor@0.15.0 downloaded                                                                                                                                                                                                                                                 
Installing builtin:serial-monitor@0.15.0...
Skipping tool configuration....
builtin:serial-monitor@0.15.0 installed
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded                                                                                                                                                                                                                                                   
Installing builtin:ctags@5.8-arduino11...
Skipping tool configuration....
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:dfu-discovery@0.1.2...
builtin:dfu-discovery@0.1.2 downloaded                                                                                                                                                                                                                                                   
Installing builtin:dfu-discovery@0.1.2...
Skipping tool configuration....
builtin:dfu-discovery@0.1.2 installed
Board Name                                 FQBN Platform ID
Adafruit Circuit Playground                     arduino:avr
Adafruit Circuit Playground Express             arduino:samd
Analog ADI                                      Arrow:samd
Arduino ATMegaNG                                arduino:avr
Arduino BT                                      arduino:avr
Arduino Diecimila                               arduino:avr
Arduino Edge Control                            arduino:mbed
Arduino Edge Control                            arduino:mbed_edge
Arduino Esplora                                 arduino:avr
Arduino Ethernet                                arduino:avr
Arduino Finder Opta                             arduino:mbed_opta
Arduino Fio                                     arduino:avr
Arduino Gemma                                   arduino:avr
Arduino Giga                                    arduino:mbed_giga
Arduino Industrial 101                          arduino:avr
Arduino Leonardo                                arduino:avr
Arduino Leonardo Ethernet                       arduino:avr
Arduino LilyPad                                 arduino:avr
Arduino LilyPad USB                             arduino:avr
Arduino M0                                      arduino:samd
Arduino M0 Pro                                  arduino:samd
Arduino MKR 1000                                arduino:samd
Arduino MKR FOX 1200                            arduino:samd
Arduino MKR GSM 1400                            arduino:samd
Arduino MKR NB 1500                             arduino:samd
Arduino MKR Vidor 4000                          arduino:samd
Arduino MKR WAN 1300                            arduino:samd
Arduino MKR WAN 1310                            arduino:samd
Arduino MKR WiFi 1010                           arduino:samd
Arduino MKR Zero                                arduino:samd
Arduino Mega                                    arduino:avr
Arduino Mega ADK                                arduino:avr
Arduino Micro                                   arduino:avr
Arduino Mini                                    arduino:avr
Arduino Nano                                    arduino:avr
[... cut ...]
Windows 10 IoT Core                             Microsoft:win10
atmega168pb-xmini                               atmel-avr-xminis:avr
atmega328p-xmini                                atmel-avr-xminis:avr
atmega328pb-xmini                               atmel-avr-xminis:avr
littleBits w6 Arduino module                    littleBits:avr
xG24 Dev Kit                                    SiliconLabs:silabs
xG24 Explorer Kit                               SiliconLabs:silabs
xG27 Dev Kit                                    SiliconLabs:silabs
```

cmaglie, 2025-04-08:
Yes, the ST platforms probably do not provide all the tools required for the platform (compiler/uploader) with a build for Raspberry PI.
If the Arduino CLI detects this situation, it will not show the platforms to prevent the installation of an incompatible platform.

By looking at the package index from ST, I see that their latest stm32 platform requires the following tools:
```
$ cat ~/.arduino15/package_stmicroelectronics_index.json | jq '.packages[].platforms[] | select(.architecture == "stm32" and .version=="2.10.1") | .toolsDependencies[]'
{
  "packager": "STMicroelectronics",
  "name": "xpack-arm-none-eabi-gcc",
  "version": "14.2.1-1.1"
}
{
  "packager": "STMicroelectronics",
  "name": "xpack-openocd",
  "version": "0.12.0-6"
}
{
  "packager": "STMicroelectronics",
  "name": "STM32Tools",
  "version": "2.3.0"
}
{
  "packager": "STMicroelectronics",
  "name": "CMSIS",
  "version": "5.9.0"
}
{
  "packager": "STMicroelectronics",
  "name": "STM32_SVD",
  "version": "1.18.0"
}
```

looking for the tools specific for the Raspberry PI (the ones with the property `"host"=="arm-linux-gnueabihf"`) I see that all the tools have the build for ARM except `STM32Tools`:
```
$ cat ~/.arduino15/package_stmicroelectronics_index.json | jq '.packages[].tools[] | select(.name == "xpack-arm-none-eabi-gcc" and .version=="14.2.1-1.1").systems[] | select(.host == "arm-linux-gnueabihf")'
{
  "host": "arm-linux-gnueabihf",
  "url": "https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v14.2.1-1.1/xpack-arm-none-eabi-gcc-14.2.1-1.1-linux-arm.tar.gz",
  "archiveFileName": "xpack-arm-none-eabi-gcc-14.2.1-1.1-linux-arm.tar.gz",
  "checksum": "SHA-256:7a0beb722c4a290e35d041d942612a38336d76a60c78286d116e79e82d82f285",
  "size": "271654556"
}
$ cat ~/.arduino15/package_stmicroelectronics_index.json | jq '.packages[].tools[] | select(.name == "xpack-openocd" and .version=="0.12.0-6").systems[] | select(.host == "arm-linux-gnueabihf")'
{
  "host": "arm-linux-gnueabihf",
  "url": "https://github.com/xpack-dev-tools/openocd-xpack/releases/download/v0.12.0-6/xpack-openocd-0.12.0-6-linux-arm.tar.gz",
  "archiveFileName": "xpack-openocd-0.12.0-6-linux-arm.tar.gz",
  "checksum": "SHA-256:ff7ba12cc72f46b686d249dbe1131cc0e8e7ffeffbcc06d2290f0db203d412f3",
  "size": "2527318"
}
$ cat ~/.arduino15/package_stmicroelectronics_index.json | jq '.packages[].tools[] | select(.name == "CMSIS" and .version=="5.9.0").systems[] | select(.host == "arm-linux-gnueabihf")'
{
  "host": "arm-linux-gnueabihf",
  "url": "https://github.com/stm32duino/ArduinoModule-CMSIS/releases/download/5.9.0/CMSIS-5.9.0.tar.bz2",
  "archiveFileName": "CMSIS-5.9.0.tar.bz2",
  "checksum": "SHA-256:693aabf1a793512316bfc9809203a5cfe304931e981ba708b51419939c3cefcd",
  "size": "2365397"
}
$ cat ~/.arduino15/package_stmicroelectronics_index.json | jq '.packages[].tools[] | select(.name == "STM32_SVD" and .version=="1.18.0").systems[] | select(.host == "arm-linux-gnueabihf")'
{
  "host": "arm-linux-gnueabihf",
  "url": "https://github.com/stm32duino/stm32_svd/releases/download/1.18.0/STM32_SVD-1.18.0.tar.bz2",
  "archiveFileName": "STM32_SVD-1.18.0.tar.bz2",
  "checksum": "SHA-256:a4e5b0f8550959d744bcc9b22a31dce87929e491631d4d7724728eca69b4dfac",
  "size": "13633290"
}
$ cat ~/.arduino15/package_stmicroelectronics_index.json | jq '.packages[].tools[] | select(.name == "STM32Tools" and .version=="2.3.0").systems[] | select(.host == "arm-linux-gnueabihf")'
$
```

So basically, the platform does not provide an `STM32Tools` build for the Raspberry PI, the Arduino CLI is working as expected.
You may want to ask the developers of the STM32 package if they can add such a build and update the index, considering their availability to do so.

omicronns, 2025-04-11:
Thank you for explanation. Actually I didn't want to develop on the Raspberry PI, but I wanted to use `arduino-cli` to program STM32 based via arduino bootloader with RPi. Is it possible to do without those tools? It seems that you have to provide board type fqbn to do so even if you want to program a binary file compiled somewhere else, and I'm not sure but I would expect, that STM32 specific tools should not be required for that.

cmaglie, 2025-04-18:
We know very little about 3rd party platforms and the solutions they use. The Arduino CLI knows that the tools listed in the platform index are required for the platform to work properly; some may be needed for upload (with or without bootloader), some others for compilation, but there is no easy general way to determine which one is required for each purpose.

If you want to give it a try, you may try to remove the following entry from the STM package index `~/.arduino15/package_stmicroelectronics_index.json` (around line 937):
```diff
               "version": "0.12.0-6"
             },
-            {
-              "packager": "STMicroelectronics",
-              "name": "STM32Tools",
-              "version": "2.3.0"
-            },
             {
               "packager": "STMicroelectronics",
```
And try to install the platform; this way, the Arduino CLI should not try to install STM32Tool, but I repeat, it's a blind shot, it may work or it may not.

closed 2025-04-25 by omicronns