arduino/arduino-cli #2888: `core install` command ignores `network.proxy` configuration

Dmitro25, 2025-04-13:
### Describe the problem

Arduino CLI must access the Internet for various operations. It can be configured to use a proxy server via the [`network.proxy` configuration key](https://arduino.github.io/arduino-cli/dev/configuration/#:~:text=the%20network%20connection.-,proxy,-%2D%20URL%20of%20the).

🐛 The `network.proxy` configuration key is ignored by the `core install` command.

### To reproduce

#### Setup

```text
$ git clone https://github.com/arduino/arduino-cli && cd arduino-cli

[...]

$ git checkout 6320f6eb  # Checkout the last good commit.

[...]

$ task build

[...]

$ export ARDUINO_DIRECTORIES_DATA="/tmp/arduino-cli-directories/data"  # Use a throwaway data folder for the demo.

$ export ARDUINO_DIRECTORIES_DOWNLOADS="/tmp/arduino-cli-directories/downloads"

$ rm -r "$ARDUINO_DIRECTORIES_DATA" "$ARDUINO_DIRECTORIES_DOWNLOADS"  # Make sure there aren't existing files that would throw off the results.

[...]

$ ./arduino-cli lib update-index  # This is required as a preparatory operation because `core install` unnecessarily triggers a libraries index update if it is not already updated.

$ export ARDUINO_NETWORK_PROXY="http://foouser:foopass@httpstat.us/404:42/"  # Configure Arduino CLI to use a proxy, intentionally using a non-functional configuration.
```

#### Demonstration of correct operation pre-regression

```text
$ ./arduino-cli core install arduino:avr  # Attempt to install a platform while using a proxy.

Downloading packages...
arduino:arduinoOTA@1.3.0 Get "https://downloads.arduino.cc/tools/arduinoOTA-1.3.0-windows_386.zip": Internal Server Error
Error during install: Get "https://downloads.arduino.cc/tools/arduinoOTA-1.3.0-windows_386.zip": Internal Server Error
```

🙂 The operation failed due to the non-functional proxy configuration, proving that the `core install` command does use the configured proxy for its downloads.

#### Demonstration of incorrect operation post-regression

```text
$ git checkout master

[...]

$ task build

[...]

$ ./arduino-cli core install arduino:avr

$ ./arduino-cli core install arduino:avr
Downloading packages...
arduino:arduinoOTA@1.3.0 downloaded
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 downloaded                                                                                                                                                                                                                  
arduino:avrdude@6.3.0-arduino17 downloaded
arduino:avr@1.8.6 downloaded
Installing arduino:arduinoOTA@1.3.0...
Configuring tool....
arduino:arduinoOTA@1.3.0 installed
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7...
Configuring tool....
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 installed
Installing arduino:avrdude@6.3.0-arduino17...
Configuring tool....
arduino:avrdude@6.3.0-arduino17 installed
Installing platform arduino:avr@1.8.6...
Configuring platform....
Platform arduino:avr@1.8.6 installed
```

🐛 The installation was successful even though Arduino CLI was configured to use a non-functional proxy. This proves that the `core install` command does not use the configured proxy for its downloads.

#### Demonstration of correct operation of `lib install` command

```text
$ export ARDUINO_DIRECTORIES_USER="/tmp/arduino-cli-directories/user"

$ rm -r "$ARDUINO_DIRECTORIES_USER"  # Make sure there aren't existing files that would throw off the results.

[...]

```text
$ ./arduino-cli lib install Servo

Downloading Servo@1.2.2...
Servo@1.2.2 Get "https://downloads.arduino.cc/libraries/github.com/arduino-libraries/Servo-1.2.2.zip?query=install": Internal Server Error
Error installing Servo: Can't download library: Get "https://downloads.arduino.cc/libraries/github.com/arduino-libraries/Servo-1.2.2.zip?query=install": Internal Server Error
```

🙂 The operation failed due to the non-functional proxy configuration, proving that the `lib install` command does use the configured proxy for its downloads.

### Expected behavior

All operations that access the Internet respect the `network.proxy` configuration.

### Arduino CLI version

23ed1b12f7263b29b2264ffa9944b8f804d57c6f

### Operating system

Windows

### Operating system version

- 10, build 1809
- 11

### Additional context

I bisected the regression to 1fddba76a15360b083f457c1028065f764747b10 / https://github.com/arduino/arduino-cli/pull/2565 (does not occur when using the build at the previous commit 6320f6eb).

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

alessio-perugini, 2025-04-22:
@Dmitro25 we have released the fix in https://github.com/arduino/arduino-cli/releases/tag/v1.2.2

Dmitro25, 2025-04-22:
Thank you. It works!