arduino/arduino-cli #2893: Make release binary name fixed.

dojyorin, 2025-04-19:
### Describe the request

I think it would be better to remove version number from release binary name and make it a fixed name.

The advantage of this is that if you want to get latest release for example, you can get it with a single link like this:

```sh
curl https://github.com/arduino/arduino-cli/releases/latest/download/arduino-cli_Linux_64bit.tar.gz
```

Currently, because version name is included, two steps are required: first obtaining download url of latest release using GitHub API etc, and then accessing binary.

```sh
curl $(curl https://api.github.com/repos/arduino/arduino-cli/releases/latest | yq '.assets[] | select(.name == "*Linux_64bit.zip").browser_download_url')
```

```txt
arduino-cli_1.2.0_Linux_64bit.tar.gz
            ^^^^^
```

### Describe the current behavior

Release binaries contain version number.

### Arduino CLI version

1.2.0

### Operating system

N/A

### Operating system version

-

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My request contains all necessary details

kittaakos, 2025-04-26:
> The advantage of this is that if you want to get latest release for example, you can get it with a single link like this:

> Currently, because version number is included, two steps are required: first obtaining download url of latest release using GitHub API etc, and then accessing binary.

Hi, just chiming in 👋 Great idea! I know your pain. It does not solve your problem, but I always get the Arduino resources (CLI, language server, etc. ) from the official Arduino download servers. They're just superior to GH release assets.

I would not change the release artifacts name, though, but rather ask the Arduino people to redirect the no version to the latest version on their official download servers; there was something like this in the past with the nightly Arduino IDE; it redirected to the actual latest nightly. (https://github.com/arduino/arduino-ide/blob/45f06718b4b1ca1d5917b1b68f45244072f11734/README.md?plain=1#L45-L48)

dojyorin, 2025-05-03:
Thank you for idea. I will try it :)