arduino/arduino-cli #2899: Consitent `Client.Timeout or context cancellation while reading body` error when installing a core with `1.2.2`

kittaakos, 2025-04-26:
### Describe the problem

I suspect a regression with `1.2.2`, whenever I install a core, I randomly get a `Client.Timeout`. I do not have the best Internet connection but it was always like that, and I could reproduce it with `1.2.0`, for example, or earlier versions.

I did some local testing and switched between `1.2.0` and `1.2.2` CLI versions and tried to install the ESP32 core after wiping `~/Users/kittaakos/Library/Arduino15/staging`. It always failed for me with `1.2.2`. There were no issues with `1.2.0`. I tried it several times. This could be a temporary connection issue on my side. If so, please go ahead with closing this issue, but I bet a 3rd party dependency update could also cause it.

### To reproduce

```
./arduino-cli core update-index --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
```

```
./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
```

With `1.2.2`:
```
./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
Tool esp32:xtensa-esp-elf-gdb@14.2_20240403 already installed
Tool esp32:riscv32-esp-elf-gdb@14.2_20240403 already installed
Tool esp32:openocd-esp32@v0.12.0-esp32-20241016 already installed
Tool esp32:esptool_py@4.9.dev3 already installed
Tool esp32:mkspiffs@0.2.3 already installed
Tool esp32:mklittlefs@3.0.0-gnu12-dc7f933 already installed
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages...
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 already downloaded
esp32:esp-x32@2411 esp32:esp-x32@2411 already downloaded
esp32:esp-rv32@2411 net/http: request canceled (Client.Timeout or context cancellation while reading body)
Error during install: net/http: request canceled (Client.Timeout or context cancellation while reading body)
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % ./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
Tool esp32:xtensa-esp-elf-gdb@14.2_20240403 already installed
Tool esp32:riscv32-esp-elf-gdb@14.2_20240403 already installed
Tool esp32:openocd-esp32@v0.12.0-esp32-20241016 already installed
Tool esp32:esptool_py@4.9.dev3 already installed
Tool esp32:mkspiffs@0.2.3 already installed
Tool esp32:mklittlefs@3.0.0-gnu12-dc7f933 already installed
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages...
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 already downloaded
esp32:esp-x32@2411 esp32:esp-x32@2411 already downloaded
esp32:esp-rv32@2411 context deadline exceeded (Client.Timeout or context cancellation while reading body)
Error during install: context deadline exceeded (Client.Timeout or context cancellation while reading body
```

OR

```
./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages...
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 net/http: request canceled (Client.Timeout or context cancellation while reading body)
Error during install: net/http: request canceled (Client.Timeout or context cancellation while reading body)
```

----

I could not reproduce it with `1.2.0`:

```
./arduino-cli version
arduino-cli  Version: 1.2.0 Commit: 9c495211 Date: 2025-02-24T15:57:28Z
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % ./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages...
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 downloaded                                         
esp32:esp-x32@2411 downloaded                                                                            
esp32:xtensa-esp-elf-gdb@14.2_20240403 downloaded                                                        
esp32:esp-rv32@2411 downloaded                                                                           
esp32:riscv32-esp-elf-gdb@14.2_20240403 downloaded                                                       
esp32:openocd-esp32@v0.12.0-esp32-20241016 downloaded                                                    
esp32:esptool_py@4.9.dev3 downloaded                                                                     
esp32:mkspiffs@0.2.3 downloaded                                                                          
esp32:mklittlefs@3.0.0-gnu12-dc7f933 downloaded                                                          
esp32:esp32@3.2.0 downloaded                                                                             
Installing esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1...
Configuring tool....
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 installed
Installing esp32:esp-x32@2411...
Configuring tool....
esp32:esp-x32@2411 installed
Installing esp32:xtensa-esp-elf-gdb@14.2_20240403...
Configuring tool....
esp32:xtensa-esp-elf-gdb@14.2_20240403 installed
Installing esp32:esp-rv32@2411...
Configuring tool....
esp32:esp-rv32@2411 installed
Installing esp32:riscv32-esp-elf-gdb@14.2_20240403...
Configuring tool....
esp32:riscv32-esp-elf-gdb@14.2_20240403 installed
Installing esp32:openocd-esp32@v0.12.0-esp32-20241016...
Configuring tool....
esp32:openocd-esp32@v0.12.0-esp32-20241016 installed
Installing esp32:esptool_py@4.9.dev3...
Configuring tool....
esp32:esptool_py@4.9.dev3 installed
Installing esp32:mkspiffs@0.2.3...
Configuring tool....
esp32:mkspiffs@0.2.3 installed
Installing esp32:mklittlefs@3.0.0-gnu12-dc7f933...
Configuring tool....
esp32:mklittlefs@3.0.0-gnu12-dc7f933 installed
Installing platform esp32:esp32@3.2.0...
Configuring platform....
Platform esp32:esp32@3.2.0 installed
```

### Expected behavior

The `1.2.0` behavior

### Arduino CLI version

1.2.2

### Operating system

macOS

### Operating system version

15.4.1

### Additional context

I have not tried the nightly because the release tag is the HEAD of the `master` (c11b9dd5), therefore, it must be the same as the nightly:

```sh
./arduino-cli version
arduino-cli  Version: 1.2.2 Commit: c11b9dd5 Date: 2025-04-22T13:51:07Z
```

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

AlainF-ing, 2025-04-27:
I have the same issue with ESP32 core only.
I use 2 other cores ESP8266 and Rp2040  installed using script  at the same time without any error.
I have this issue on Bookworm (latest) on Raspberry Pi 4B
I have tried my script on Fedora Workstation 42 Beta same error only 2 cores installed without timeout. ESP32 core can not install.

alessio-perugini, 2025-04-28:
I suspect it is related to the fix https://github.com/arduino/arduino-cli/pull/2889 that correctly honors the patch done in v1.1.0 (https://github.com/arduino/arduino-cli/pull/2770)

Can you try it by increasing the default timeout from 60s to 180s?
`arduino-cli config set network.connection_timeout 180s`

If this is confirmed, we should probably increase the default timeout

kittaakos, 2025-04-28:
Thank you!

> If this is confirmed

Confirmed. Increasing the timeout works.

```
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % rm -rf ~/Library/Arduino15/packages/esp32 
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % rm -rf ~/Library/Arduino15/staging       
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % ./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages...
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 downloaded                                                                            
esp32:esp-x32@2411 downloaded                                                                                                               
esp32:xtensa-esp-elf-gdb@14.2_20240403 downloaded                                                                                           
esp32:esp-rv32@2411 net/http: request canceled (Client.Timeout or context cancellation while reading body)                                  
Error during install: net/http: request canceled (Client.Timeout or context cancellation while reading body)
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % rm -rf ~/Library/Arduino15/staging
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % rm -rf ~/Library/Arduino15/packages/esp32
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % ./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % ./arduino-cli config set network.connection_timeout 180s
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % ./arduino-cli config dump                               
network:
    connection_timeout: 180s

kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % ./arduino-cli core install esp32:esp32 --additional-urls https://espressif.github.io/arduino-esp32/package_esp32_index.json
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages...
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 downloaded                                                                            
esp32:esp-x32@2411 downloaded                                                                                                               
esp32:xtensa-esp-elf-gdb@14.2_20240403 downloaded                                                                                           
esp32:esp-rv32@2411 downloaded                                                                                                              
esp32:riscv32-esp-elf-gdb@14.2_20240403 downloaded                                                                                          
esp32:openocd-esp32@v0.12.0-esp32-20241016 downloaded                                                                                       
esp32:esptool_py@4.9.dev3 downloaded                                                                                                        
esp32:mkspiffs@0.2.3 downloaded                                                                                                             
esp32:mklittlefs@3.0.0-gnu12-dc7f933 downloaded                                                                                             
esp32:esp32@3.2.0 downloaded                                                                                                                
Installing esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1...
Configuring tool....
esp32:esp32-arduino-libs@idf-release_v5.4-2f7dcd86-v1 installed
Installing esp32:esp-x32@2411...
Configuring tool....
esp32:esp-x32@2411 installed
Installing esp32:xtensa-esp-elf-gdb@14.2_20240403...
Configuring tool....
esp32:xtensa-esp-elf-gdb@14.2_20240403 installed
Installing esp32:esp-rv32@2411...
Configuring tool....
esp32:esp-rv32@2411 installed
Installing esp32:riscv32-esp-elf-gdb@14.2_20240403...
Configuring tool....
esp32:riscv32-esp-elf-gdb@14.2_20240403 installed
Installing esp32:openocd-esp32@v0.12.0-esp32-20241016...
Configuring tool....
esp32:openocd-esp32@v0.12.0-esp32-20241016 installed
Installing esp32:esptool_py@4.9.dev3...
Configuring tool....
esp32:esptool_py@4.9.dev3 installed
Installing esp32:mkspiffs@0.2.3...
Configuring tool....
esp32:mkspiffs@0.2.3 installed
Installing esp32:mklittlefs@3.0.0-gnu12-dc7f933...
Configuring tool....
esp32:mklittlefs@3.0.0-gnu12-dc7f933 installed
Installing platform esp32:esp32@3.2.0...
Configuring platform....
Platform esp32:esp32@3.2.0 installed
kittaakos@Akoss-MacBook-Pro-2 esp32-coredump-guide % 
```

It's not a bug, and I will close this issue.

closed 2025-04-28 by kittaakos

AlainF-ing, 2025-04-28:
Sorry guys for the delay.
This fix is working 
ESP32 core v 3.2.0 has been installed, esp32:esp-x32@2411 downloaded is 560 Mb and downloaded ok.

Thanks for your help

nuclearcat, 2025-05-24:
Hit this bug as well, i think many users with moderate connection (1-2Mb/s) will be affected by default timeout too low.

lukemorcom, 2025-09-13:
Struggled with this also while downloading the esp32 core on a VM with moderately slow albeit consistent wired internet connection, cli version 1.3.1.

Increasing timeout fixed it for me too.

bingis-khan, 2025-11-17:
This timeout approach is weird/wrong. The timeout should be from the time the last packet was received and reset while the library is actively being  downloaded.

(had the same problem, also with esp32, increasing timout helped)