arduino/arduino-cli #2936: Add `--skip-hardware` option to `core install`

dankeboy36, 2025-06-25:
### Describe the request

Please consider adding a `--skip-hardware` flag to the `core install` command to allow installation of only the toolchain and tools, excluding the hardware definition directory (e.g., `hardware/esp32/...`).
This would benefit use cases where the core is not used for compiling or uploading sketches, but the CLI tools are still required, such as decoding core dumps or flashing precompiled binaries.

#### Motivation

Currently, a `core install` always downloads the full hardware and tools payload. For example:

```
arduino-cli core install esp32:esp32
```

Creates:

```
pwd
/Users/akoskitta/Library/Arduino15/packages/esp32
```

```
tree -L 3 .
.
├── hardware
│   └── esp32
│       └── 3.1.3
└── tools
    ├── esp-rv32
    │   └── 2405
    ├── esp-x32
    │   └── 2405
    ├── esp32-arduino-libs
    │   └── idf-release_v5.3-489d7a2b-v1
    ├── esptool_py
    │   └── 4.9.dev3
    ├── mklittlefs
    │   └── 3.0.0-gnu12-dc7f933
    ├── mkspiffs
    │   └── 0.2.3
    ├── openocd-esp32
    │   └── v0.12.0-esp32-20241016
    ├── riscv32-esp-elf-gdb
    │   └── 14.2_20240403
    └── xtensa-esp-elf-gdb
        └── 14.2_20240403
```

However, the hardware definitions are not needed in workflows where the firmware is already compiled (e.g., CI builds, production flashing) or the tools are invoked manually or programmatically (e.g., esptool, gdb, mkspiffs, etc.). Skipping them could save disk space and reduce install time.

#### Proposal

Add a flag like:

```
arduino-cli core install esp32:esp32 --skip-hardware
```

This would fetch and install only the tools defined in the core’s JSON index. It would omit downloading or unpacking the `hardware/` folder.

This likely depends on how cores are structured in the index and whether the hardware and tools are independently listed. If tools can be installed standalone (as arduino-cli core install already handles tool dependencies), this might be a manageable change, but my Arduino Boards Manager knowledge is limited here, and I am just guessing.

If this is possible, would Arduino be open to such a contribution?

### Describe the current behavior

There is no option to install only the toolchain of a core.

### Arduino CLI version

1.2.0

### Operating system

macOS

### Operating system version

15.5

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My request contains all necessary details

WestfW, 2025-06-27:
That seems a bit pointless.  The "hardware" directory is a small fraction of the size of the "tools" directory (about 1.2%):
```
> pwd
/Applications/Arduino-1.8.13.app/Contents/Java/portable/packages/esp32
> du -sh *
 46M    hardware
3.9G    tools
```
(similar ratios for other platforms, assuming that they're unable to use some other platform's compiler.  Compiler toolchains are big, especially because they tend to include bulky .h files for MANY chips.)

dankeboy36, 2025-06-27:
> That seems a bit pointless.

I agree. I should have checked the ratio. Thank you

closed 2025-06-27 by dankeboy36