arduino/arduino-cli #2944: Error: 4 DEADLINE_EXCEEDED when trying to install board software

5ucur, 2025-07-06:
### Describe the problem

I'm trying to install the software needed for a RP2040 board, but the problem occurs when trying to install other boards, too. This is the error it gives (seen in the console when running the IDE from the terminal):

```
2025-07-05T22:55:54.446Z root ERROR Request install failed with error: 4 DEADLINE_EXCEEDED: context deadline exceeded (Client.Timeout or context cancellation while reading body) Error: 4 DEADLINE_EXCEEDED: context deadline exceeded (Client.Timeout or context cancellation while reading body)
    at t.callErrorFromStatus (/usr/lib/arduino-ide/lib/backend/main.js:2:921584)
    at Object.onReceiveStatus (/usr/lib/arduino-ide/lib/backend/main.js:2:949201)
    at Object.onReceiveStatus (/usr/lib/arduino-ide/lib/backend/main.js:2:942225)
    at /usr/lib/arduino-ide/lib/backend/main.js:2:1028347
    at process.processTicksAndRejections (node:internal/process/task_queues:77:11)
for call at
    at s.makeServerStreamRequest (/usr/lib/arduino-ide/lib/backend/main.js:2:948968)
    at s.platformInstall (/usr/lib/arduino-ide/lib/backend/main.js:2:1013244)
    at y.install (/usr/lib/arduino-ide/lib/backend/main.js:2:86129)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async u.onRequest (/usr/lib/arduino-ide/lib/backend/main.js:2:1265156)
    at async c.handleRequest (/usr/lib/arduino-ide/lib/backend/main.js:2:1256051)
```
I've seen similar issues over on the `arduino-cli` repo (for example, this one with [this solution](https://github.com/arduino/arduino-cli/issues/2899#issuecomment-2834199190) (increasing timeout), which is also a solution I found on the forum, too; but even at 6000s, the timeout happens within a couple minutes). However, I'm using the IDE, not the CLI, to download, and experiencing this issue. And, I have not found this problem in this repo's issues.

### To reproduce

1. Open boards manager side pane.
2. Type in 2040.
3. Select one of the three results and click its Install button.
4. Wait for a few minutes.

### Expected behavior

I would expect a successful installation; or at least if it's a timeout error, that a timeout of 6000s is more than enough.

### Arduino IDE version

The Help menu > About Arduino IDE window does not show any version numbers. Checking the Arch Linux repo, seeing as I installed just today, it's 2.3.4, which lines up with the text in the title bar.

### Operating system

Linux

### Operating system version

Arch Linux 6.12.35-1-lts

### Additional context

Unlike the stable release, the error does not happen on the nightly build.

If I missed any necessary details, please let me know. And if something is hard to understand, please ask and I will rephrase, English is not my main language.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

5ucur, 2025-07-06:
Yeah, sorry! It was late last night when I posted this. I should have realised there's no need to post as soon as I saw it working properly in the nightly. Hope the change carries over to stable soon!

per1234, 2025-07-06:
It might already be there. You were using a package that is possibly modified in unknown ways by the 3rd party package maintainer, and was certainly outdated. The latest release version is 2.3.6. There has been no meaningful development since the 2.3.6 release, so it should be functionally equivalent to the nightly. You can download the 2.3.6 release from the Arduino website:

https://www.arduino.cc/en/software/