arduino/arduino-cli #2973: ESP32 Board Manager Error

retrodorian, 2025-08-12:
### Describe the problem

Hello everyone, I've always used the Arduino IDE on all my ESP32-based boards and haven't had any problems. Currently, when I follow espressif's instructions, I keep getting this error:

```text
Downloading packages
esp32:esp32-arduino-libs@idf-release_v5.5-b66b5448-v1
esp32:esp-x32@2411
Failed to install platform: 'esp32:esp32:3.3.0'.
Error: 2 UNKNOWN: stream error: stream ID 1; PROTOCOL_ERROR; received from peer
```

I tried different ide versions and different esp32 board versions but each one gives the same error. (Yes, I added the required additional boards manager URL to preferences)

### To reproduce

1. Copy the following: 
2. Open the Arduino IDE
3. Go to Settings/Preferences
4. Paste the following url into the "Additional Boards Manager URLs" field:
   ```text
   https://espressif.github.io/arduino-esp32/package_esp32_index.json
   ```
6. Hit OK
7. Open Boards Manager.
8. Click the "Install" button on the "esp32" entry

### Expected behavior

The platform installs successfully.

### Arduino IDE version

2.3.6

### Operating system

macOS

### Operating system version

macOS 26 Tahoe Developer Beta (25A5338b)

### Additional context

I also tried it on my Windows 11 machine either and got same error message. Additionally for macOS, when I hit the nightly builds at webpage, the download does not start instead opens the home page.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [ ] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

cmaglie, 2025-08-12:
These packages are quite large. Could the issue be related to a download timeout?

retrodorian, 2025-08-12:
> These packages are quite large. Could the issue be related to a download timeout?

Yes, you are right  but I tried on 2 different machines and also switched to different networks (home router and phone hotspot with cellular data).

retrodorian, 2025-08-12:
> [@retrodorian](https://github.com/retrodorian) did you try with the [CLI 1.3.0](https://github.com/arduino/arduino-cli/releases/tag/v1.3.0)? I ask because it contains updates on the GRPC and HTTPS/SSL libraries that could indirectly fix the issue.

Thank you for fast reply. Yes I tried and gives the following error:

esp32:esp32-arduino-libs@idf-release_v5.5-b66b5448-v1 esp32:esp32-arduino-libs@idf-release_v5.5-b66b5448-v1 already downloaded
esp32:esp-x32@2411 net/http: request canceled (Client.Timeout or context cancellation while reading body)               
Error during install: net/http: request canceled (Client.Timeout or context cancellation while reading body)

cmaglie, 2025-08-12:
Ok that's due to the download timeout, try this:
```
arduino-cli config set network.connection_timeout 10m
```
or even an higher timeout if 10 minutes is not enough.