arduino/arduino-cli #3008: unable to install Boards Keep getting errors

comet424, 2025-09-18:
### Describe the problem

Installation of the "Silicon Labs" boards platform fails with the following error:

```text
Downloading packages
SiliconLabs:gcc-arm-none-eabi@12.2.rel1
Failed to install platform: 'SiliconLabs:silabs:3.0.0'.
Error: 2 UNKNOWN: stream error: stream ID 1; PROTOCOL_ERROR; received from peer
```

It downloads half and then fails.

### To reproduce

Run the following commands:

```text
$ arduino-cli config set network.connection_timeout 0  # Disable Arduino CLI's timeout mechanism to exclude it as a factor.

$ arduino-cli core install SiliconLabs:silabs

Downloading packages...
SiliconLabs:gcc-arm-none-eabi@12.2.rel1 SiliconLabs:gcc-arm-none-eabi@12.2.rel1 downloaded
SiliconLabs:openocd@0.12.0-arduino1-static downloaded                                                                                                                                                                                       
SiliconLabs:simplicitycommander@1.16.4 downloaded
SiliconLabs:silabs@3.0.0 stream error: stream ID 3; PROTOCOL_ERROR; received from peer
Error during install: stream error: stream ID 3; PROTOCOL_ERROR; received from peer
```

🐛 The download of the `SiliconLabs:silabs@3.0.0` platform archive file failed.

### Expected behavior

Platform installation is successful.

### Arduino CLI version

#### Original report

1.2.0 (version used by Arduino IDE 2.3.6)

#### Last verified with

[d5ef61d](https://github.com/arduino/arduino-cli/commit/d5ef61d674566ecf9e7015ce3ecb4df98875788e)

### Operating system

Windows

### Operating system version

11

### Additional context

The fault also occurs if I attempt to download the `SiliconLabs:gcc-arm-none-eabi@12.2.rel1` archive directly:

```text
$ wget https://github.com/SiliconLabs/arduino/releases/download/3.0.0/silabs_arduino_core-3.0.0.zst

Resolving github.com (github.com)... 140.82.114.3
Connecting to github.com (github.com)|140.82.114.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://release-assets.githubusercontent.com/github-production-release-asset/740914384/7f8eff6f-eae2-4d17-8d26-1e212db85f45?sp=r&sv=2018-11-09&sr=b&spr=https&se=2025-09-18T20%3A42%3A19Z&rscd=attachment%3B+filename%3Dsilabs_arduino_core-3.0.0.zst&rsct=application%2Foctet-stream&skoid=96c2d410-5711-43a1-aedd-ab1947aa7ab0&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skt=2025-09-18T19%3A41%3A22Z&ske=2025-09-18T20%3A42%3A19Z&sks=b&skv=2018-11-09&sig=Q3kSPjHI3z9qm%2BC44uf5etUnbV%2FGCSihL2ohGkMaSzE%3D&jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmVsZWFzZS1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1ODIyNTQ2MSwibmJmIjoxNzU4MjI1S1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1ODIyNTQ2MSwibmJmIjoxNzU4MjI1MTYxLCJwYXRoIjoicmVsZWFzZWFzc2V0cHJvZHVjdGlvbi5ibG9iLmNvcmUud2luZG93cy5uZXQifQ.qKIFIyz0ly-y6bHyyywpC4NlQ81X2r-udIVXgqJUg2c&response-content-disposition=attachment%3B%20filename%3Dsilabs_arduino_core-3.0.0.zst&response-content-type=application%2Foctet-stream [following]
--2025-09-18 12:52:41--  https://release-assets.githubusercontent.com/github-production-release-asset/740914384/7f8eff6f-eae2-4d17-8d26-1e212db85f45?sp=r&sv=2018-11-09&sr=b&spr=https&se=2025-09-18T20%3A42%3A19Z&rscd=attachment%3B+filename%3Dsilabs_arduino_core-3.0.0.zst&rsct=application%2Foctet-stream&skoid=96c2d410-5711-43a1-aedd-ab1947aa7ab0&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skt=2025-09-18T19%3A41%3A22Z&ske=2025-09-18T20%3A42%3A19Z&sks=b&skv=2018-11-09&sig=Q3kSPjHI3z9qm%2BC44uf5etUnbV%2FGCSihL2ohGkMaSzE%3D&jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmVsZWFzZS1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1ODIyNTQ2MSwibmJmIjoxNzU4MjI1MTYxLCJwYXRoIjoicmVsZWFzZWFzc2V0cHJvZHVjdGlvbi5ibG9iLmNvcmUud2luZG93cy5uZXQifQ.qKIFIyz0ly-y6bHyyywpC4NlQ81X2r-udIVXgqJUg2c&response-content-disposition=attachment%3B%20filename%3Dsilabs_arduino_core-3.0.0.zst&response-content-type=application%2Foctet-stream
Resolving release-assets.githubusercontent.com (release-assets.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to release-assets.githubusercontent.com (release-assets.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 602042581 (574M) [application/octet-stream]
Saving to: 'silabs_arduino_core-3.0.0.zst'

silabs_arduino_core-3.0.0.zst                               43%[==========================================================>                                                                              ] 250.00M   789KB/s    in 5m 2s

2025-09-18 12:57:43 (849 KB/s) - Connection closed at byte 262144000. Retrying.

--2025-09-18 12:57:44--  (try: 2)  https://release-assets.githubusercontent.com/github-production-release-asset/740914384/7f8eff6f-eae2-4d17-8d26-1e212db85f45?sp=r&sv=2018-11-09&sr=b&spr=https&se=2025-09-18T20%3A42%3A19Z&rscd=attachment%3B+filename%3Dsilabs_arduino_core-3.0.0.zst&rsct=application%2Foctet-stream&skoid=96c2d410-5711-43a1-aedd-ab1947aa7ab0&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skt=2025-09-18T19%3A41%3A22Z&ske=2025-09-18T20%3A42%3A19Z&sks=b&skv=2018-11-09&sig=Q3kSPjHI3z9qm%2BC44uf5etUnbV%2FGCSihL2ohGkMaSzE%3D&jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmVsZWFzZS1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1ODIyNTQ2MSwibmJmIjoxNzU4MjI1MTYxLCJwYXRoIjoicmVsZWFzZWFzc2V0cHJvZHVjdGlvbi5ibG9iLmNvcmUud2luZG93cy5uZXQifQ.qKIFIyz0ly-y6bHyyywpC4NlQ81X2r-udIVXgqJUg2c&response-content-disposition=attachment%3B%20filename%3Dsilabs_arduino_core-3.0.0.zst&response-content-type=application%2Foctet-stream
Connecting to release-assets.githubusercontent.com (release-assets.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 618 jwt:expired
2025-09-18 12:57:45 ERROR 618: jwt:expired.
```

#### Additional reports

- https://forum.arduino.cc/t/cannot-install-stm32-nor-esp32-boards-to-arduino-ide-1-8-19/1405156
- https://forum.arduino.cc/t/espressif-board-manager-3-3-0-fails/1405287
- https://forum.arduino.cc/t/unable-to-install-stm32-based-boards/1406610

#### Related

- https://github.com/arduino/arduino-cli/issues/2973

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

comet424, 2025-09-18:
@per1234 hi ok ya there is no way i can download in 5 min.. that 600 meg file when you click on that link  took 30 min or so  my internet is only 3 megabit   i live in the country  so we cant get high speed.. no 1gb  i just get 3 megabit if i lucky 

so  i did download there file and i placed it in the one folder  but  i followed youtube video but it didnt work  where you place it in the 
C:\Users\<username>\AppData\Local\Arduino15\staging\packages

but it did nothing  and apparently i posted that information i copied it but it seems to be deleted from my post... where i can download it from  Github but i also use sometimes  Download with the Internet Download Manager

also my comment i installed  ide 2.2.1  and then 2.3.6 went missing too  i had done what the videos said to fix the issue.. some how all my text went missing  that i tried

i had followed the guys video  but here is what i did

[image]
[image]

but you mentioned the folder goes into the staging packages..  did i even download the right file too?  as its 600 meg zipped unzipped that folder is 1.84gb    so i just put it in the staging location..  
but ya this is bs  as i had to use the Internet Download Manager to download it...  so that means  like Home Asistant that downloads from Github too  that means my home assistant will have issues too?

per1234, 2025-09-18:
> but did i download the right file?

No, you did not. This is not an appropriate place to provide support. Please post over on Arduino Forum. I will see your post there and will be happy to provide support:

https://forum.arduino.cc/c/development-tools/ide-2-x/93

>my article is all missing

I had to rewrite your issue because it was not of sufficient quality to allow the developers to efficiently evaluate and act on it. Please make a stronger effort in the future.

comet424, 2025-09-18:
ah here i thought i did ok that i explained things right..

if you happened to be a an admin on the Arduino forum i tried to post on there .. but my username is reserved  cuz it was deleted for in activitity  and i cant re use my email with my username

guess ill try to make a new account.. but ok ill post over there