arduino/arduino-cli #3010: Monitor: "configure failed: invalid argument" when selecting baudrate returned by EnumerateMonitorPortSettingsResponse

dankeboy36, 2025-09-20:
### Describe the problem

When trying to change the baudrate, I get this error:

```json
{"error": "command 'configure' failed: invalid argument"}
```

This happens when I pick a baudrate value returned by [`EnumerateMonitorPortSettings`](https://arduino.github.io/arduino-cli/1.3/rpc/commands/#enumeratemonitorportsettingsresponse).

### To reproduce

Start the Arduino CLI in daemon mode:

```
./arduino-cli version
arduino-cli  Version: git-snapshot Commit: d5ef61d6 Date: 2025-09-20T13:04:53Z

./arduino-cli daemon
Daemon is now listening on 127.0.0.1:50051
{"IP":"127.0.0.1","Port":"50051"}
```

Open a new shell inside the `arduino-cli` repo and run the following:

Generate the .proto files:

```
buf export . -o ./tmp/out
```

Create:

```
grpcurl \
  -plaintext \
  -import-path ./tmp/out \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Create
```
```json
{
  "instance": {
    "id": 1
  }
}
```

Init:

```
grpcurl \
  -plaintext \
  -import-path ./tmp/out \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance": {"id": 1}}' \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Init
```

Enumerate monitor port settings (FQBN is optional):

```
grpcurl -plaintext \
  -import-path ./tmp/out \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d '{"instance":{"id":1}, "port_protocol":"serial", "fqbn":"esp32:esp32:esp32da"}' \
  localhost:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService/EnumerateMonitorPortSettings
```

```json
{
  "settings": [
    {
      "settingId": "stop_bits",
      "label": "Stop bits",
      "type": "enum",
      "enumValues": [
        "1",
        "1.5",
        "2"
      ],
      "value": "1"
    },
    {
      "settingId": "baudrate",
      "label": "Baudrate",
      "type": "enum",
      "enumValues": [
        "300",
        "600",
        "750",
        "1200",
        "2400",
        "4800",
        "9600",
        "19200",
        "31250",
        "38400",
        "57600",
        "74880",
        "115200",
        "230400",
        "250000",
        "460800",
        "500000",
        "921600",
        "1000000",
        "2000000"
      ],
      "value": "9600"
    },
    {
      "settingId": "bits",
      "label": "Data bits",
      "type": "enum",
      "enumValues": [
        "5",
        "6",
        "7",
        "8",
        "9"
      ],
      "value": "8"
    },
    {
      "settingId": "dtr",
      "label": "DTR",
      "type": "enum",
      "enumValues": [
        "on",
        "off"
      ],
      "value": "off"
    },
    {
      "settingId": "parity",
      "label": "Parity",
      "type": "enum",
      "enumValues": [
        "none",
        "even",
        "odd",
        "mark",
        "space"
      ],
      "value": "none"
    },
    {
      "settingId": "rts",
      "label": "RTS",
      "type": "enum",
      "enumValues": [
        "on",
        "off"
      ],
      "value": "off"
    }
  ]
}
```

This lists the supported settings, including baudrates like 750.

Create the monitor duplex:

```
grpcurl \
  -plaintext \
  -import-path ./tmp/out \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -d @ \
  127.0.0.1:50051 \
  cc.arduino.cli.commands.v1.ArduinoCoreService.Monitor
```

Open the monitor:

```json
{
  "openRequest": {
    "instance": { "id": 1 },
    "fqbn": "esp32:esp32:esp32da",
    "port": {
      "address": "/dev/cu.usbserial-0001",
      "protocol": "serial"
    },
    "portConfiguration": {
      "settings": [
        { "settingId": "baudrate", "value": "74880" }
      ]
    }
  }
}
```

This works and data comes in from the board.

```jsonc
{
  "success": true
}
{
  "rxData": "KqtQqlRVC1WlC6XVVbVVolRFsBHPIse4gR7DYSADIzAB/g=="
}
{
  "rxData": "MA=="
}
{
  "rxData": "LjAyMDAJMA=="
}
// ...
```

But if I try to change the baudrate to 750 (which was listed as valid):

```json
{
  "updatedConfiguration": {
    "settings": [
      { "settingId": "baudrate", "value": "750" }
    ]
  }
}
```

I get this error:

```json
{
  "error": "command 'configure' failed: invalid argument"
}
```

### Expected behavior

 - If a baudrate is not actually supported, it should not appear in the list.
 - Or, if it cannot be detected reliably, the error should be more specific so clients can handle it.

### Arduino CLI version

d5ef61d6

### Operating system

macOS

### Operating system version

15.6.1

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [x] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-09-20:
Hi @dankeboy36. Are you able to reproduce the fault reliably? I ask because the first time I tried it, I did encounter the same problem, but then every other time I tried it worked as expected. I have no idea what I might have done differently the first time.

Does the fault only occur when using `750` specifically, or does it occur with any baud rate? I ask because the time I was able to reproduce it, I found I was able to reproduce it with any baud rate.

dankeboy36, 2025-09-20:
Thank you for your time. You were right, it's not happening all the time, but I can reliably reproduce it when I start the monitor connection with a 74880 baudrate, and then select the 750. I did not get the error with other baudrates. I've refined the issue description, and  I can check if I can reproduce it on another Mac if you need it.